bigtreetech/BIGTREETECH-TouchScreenFirmware #546: Cannot change Marlin Mode text and background color

blair9095, 2020-04-08:
I am unable to change the background or text color when in the Marlin Mode.  I have changed the color of the touch screen (in firmware) but the desired changes to the Marlin screen  don't work

blair9095, 2020-04-08:
Clarification:  all changes made to latest firmware using latest version of BIGTREETECH-TouchScreenFirmware-master downloaded on 4/7.

bigtreetech, 2020-04-10:
Hello, Which TFT do you use? I just tested and it works normally. The last two button in the "Settings - > Screen" menu is the background color and font color in Marlin(LCD 12864) mode

blair9095, 2020-04-10:
I can change the Marlin colors using 
Touch>Menu>Setting>Screen>Foreground or Background
#define ST7920_BKCOLOR GRAY
#define ST7920_FNCOLOR BLACK
but it still comes up Black on Yellow.

closed 2020-08-19 by bigtreetech