bigtreetech/BIGTREETECH-TouchScreenFirmware #548: Scrambled Marlin Screen

hfog, 2020-04-08:
### Description

Marlin screen becomes scrambled over time. 

### Steps to reproduce

1. Print something with the TFT on the marlin screen. Longer the better.
2. Wait and watch.
3. Problem is often exacerbated (manifests quicker) if you are working through the menus and rapidly spinning the dial but it will eventually happen all on its own even if you aren't navigating the menus at all.

**Expected behavior**
Screen should always look correct no matter what I'm doing.

**Actual behavior**
Screen gets scrambled. 
The minor manifestation of this bug is an offset and wrapped screen (see below) but I've also had the screen completely messed up with only a few lines of the screen updating and wrapping over themselves.

### Hardware Variant
I'm using a BTT-TFT35-E3-V3.0 connected to a BIGTREETECH-SKR-mini-E3 running on an Ender 3 PRO.

### TFT Firmware Version & Main Board Firmware details
Latest TFT firmware pulled from repo as of April 5 but I've been seeing this for several versions now.
Marlin version 2.0.5.3 (but I've also seen it on earlier versions)

### Additional Information
This does NOT happen with the stock Ender 3 Pro screen. I've flipped back and forth several times to verify this. It is DEFINITELY something to do with the TFT.

[image]

hfog, 2020-04-08:
I should add that this only affects the emulator screen. I've never seen it do this on the touch menus.

radek8, 2020-04-08:
I can confirm the same behavior.
TFT 35-E3-V3.0.25.3 - Bigtreetech SKR 1.4

thisiskeithb, 2020-04-09:
@bigtreetech: Is there a set of optimal `ST7920_DELAY_*` settings for "Marlin Mode"?

Example:
```cpp
#define ST7920_DELAY_1 DELAY_NS(150)
#define ST7920_DELAY_2 DELAY_NS(150)
#define ST7920_DELAY_3 DELAY_NS(150)
```

jjwagner, 2020-04-09:
Confirmed with BTT-TFT24 1.1 connected to a BIGTREETECH-SKR-mini-E3 1.2 on an XVico X3 (Ender 3 clone). I had to make my own cable to connect the SKR to the TFT24's EXP1 and EXP2 port so I could use the stock Marlin interface.

Have not tried with a generic RepRapDiscount Full Graphic display. I'll try that tonight.

I've been chasing this for a while. Something happens to the board and display during long prints. The TFT24 emulated screen gets scrambpled and the X-axis TMC2209 loses some of its settings (verified with M122), which doubles the scale on the X-axis and causes it to crash into the right side of the carriage.

radek8, 2020-04-09:
I use Marlin
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

hfog, 2020-04-09:
This looks very much like something is overwriting memory that controls how the screen is laid out during printing. Looks like there's probably a register somewhere that determines where the upper right corner of the screen is and that is getting overwritten by something else.
I've looked through the code but so far I haven't been able to identify it and since there's not really a debugging facility this could prove pretty hard to nail down.

thisiskeithb, 2020-04-09:
When this happens on a non-emulated full graphic/12864 LCD, it's due to incorrect timings. That's why I asked BigTreeTech if there were a specific set of timings that were recommended vs. Marlin's defaults.

radek8, 2020-04-09:
The problem occurs in fullscreen and in emulated mode.
When connected LCD 12864 displays correctly.
The TFT display does not display correctly. It must restart

[image]
[image]

reQu1em00, 2020-04-11:
I get the same issue.  BTT e3 mini v1.2 and TFT35.  Often happens if you spin through the interface quickly.

hfog, 2020-04-14:
Bumping the delays does seem to help a bit but I still haven't been able to get rid of the problem entirely despite trying any number of combinations. 
Right now I'm running these settings which seem quite extreme:
     #define ST7920_DELAY_1 DELAY_NS(100) // After CLK LOW
     #define ST7920_DELAY_2 DELAY_NS(900) // After DAT
     #define ST7920_DELAY_3 DELAY_NS(100) // After CLK HIGH

jjwagner, 2020-04-14:
Can confirm @radek8 - I attached a RRD Full Graphic controller to my SKR Mini E3 and it did not get scrambled at all during a ~3 hour long print. The emulated display in the TFT24 always ends up scrambled in a print that long, using the exact same cable.

Mikouille, 2020-05-23:
Hi everyone, i have same trouble with a TFT24 configured in full screen when it's in emulated graphical mode and a SKR Pro V1.1. Does anyone know how to fix it ?

c3D-Dan, 2020-07-03:
Same issue here

closed 2020-11-03 by stale[bot]