bigtreetech/BIGTREETECH-TouchScreenFirmware #551: [FR] Added command M420 T0 V1

radek8, 2020-04-08:
Please complete command M420 T0 V1
For example, under an empty button in a terminal

See picture
[image]
[image]
[image]
[SendGcode.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4452888/SendGcode.zip)
[image]
[image]

guruathwal, 2020-04-09:
These types of g-code are added in custom g-codes menu.

radek8, 2020-04-09:
does not appear in custom G-codes
full screen, only last line

closed 2020-05-11 by radek8