bigtreetech/BIGTREETECH-TouchScreenFirmware #552: [BUG][FR] Repetier-Firmware compatibility

GenaUser, 2020-04-09:
On my printers, I use Repetier-Firmware with TFT35 V3 on Rumba  and Ramps boards.
LCD screen worked perfectly until 1 month ago. Since update March 14, I can't use my TFT in touch screen mode because it can't connect to the board « No printer attached ».
I made a rapid test with a not configured marlin and the connexion is on.
Do you think you can restore compatibility with Repetier, or all users of this firmware should abandon BigTreeTech screens?

AntoszHUN, 2020-04-09:
Hi! I last used Repetier firmware 3 years ago. Unfortunately, it is not evolving anywhere. Very backward. I suggest you switch to Marlin firmware. You will not regret it. ;)

GenaUser, 2020-04-09:
Hi @AntoszHUN 
I don't totally agree with you
1) Repetier-Firmware V2 (32 bits) is in progress https://github.com/repetier/Repetier-Firmware/tree/dev2
2) All Repetier users don't have the possibility / informations to configure a new firmware.
3) Personally, I can't use Marlin on my printers because only Repetier allows the use of extra motors with endstop, which are essential for switch head.
To better understand, here is a hold print test with my printer: https://www.youtube.com/watch?v=I6jBb85iDac&t=15s

bigtreetech, 2020-04-10:
will test now,  thanks for your feedback！

GenaUser, 2020-04-10:
@bigtreetech thank you for working to restore this compatibility!

bigtreetech, 2020-04-11:
Hello, it should be fixed in #563, you can test it

GenaUser, 2020-04-11:
@bigtreetech I did a test.
Connection is on after firmware uploaded, but after reboot, the connection is definitely lost.
If I switch on Marlin/Repetier mode and I back to touch screen mode, the connection is restored

You are close to the solution.

GenaUser, 2020-04-17:
Please @bigtreetech, could you fix the last bug so that Repetier users can use your screens correctly again (connection lost after reboot).
Thanks in advance

AntoszHUN, 2020-04-18:
Of course. I understand. You're obviously right. But Repetier firmware is more recommended for Delta printers. Several years ago when I started 3D printing, Repetier was very lagging behind Marlin. It may have evolved since then. For example, I only use the Repetier Host program from a slicer. It’s mature and I’ve never had a problem with it. But I'm sure Marlin knows what you need. Indeed. Even more.

bigtreetech, 2020-04-20:
@GenaUser will test again.

GenaUser, 2020-04-20:
Thank you @bigtreetech, I look forward to this change.
Out of curiosity I tested today's version, the problem is still there.

AntoszHUN, 2020-04-28:
Okay. Pls close.

GenaUser, 2020-05-02:
@AntoszHUN  I can't close because the bug is not solved. 
I may have expressed myself badly. I thanked because @bigtreetech had to keep working on this problem, not because it's solved.
Repetier users are still waiting to be able to use these screens properly.
Please @bigtreetech, could you add this last bug to your to-do list? 
_"Connection is on after firmware uploaded, but after reboot, the connection is lost.
If I switch on Marlin/Repetier mode and I back to touch screen mode, the connection is restored, until reboot."_

Thanks in advance

AntoszHUN, 2020-05-02:
All right. The problem is interesting.

closed 2020-05-27 by GenaUser