bigtreetech/BIGTREETECH-TouchScreenFirmware #554: [BUG] Not work save buton in the BabyStep and ProbeOffset

radek8, 2020-04-09:
294/5000
The Save button does not work in the BabyStep and ProbeOffset menu.
after commenting the "if" condition (in ProbeOffset.c, BabyStep.c) the button is functional.
Please repair

       case KEY_ICON_4:
         //if(infoMachineSettings.EEPROM == 1) {
            storeCmd ("M500 \ n");
         //}
         break;

My configuration
Bigtreetech TFT35 E V3,0.25.3
Bigtreetech SKR 1,4 - Marlin 2.0.5.2

PPacione, 2020-04-10:
Doing the above will give you a message that the change is saved but it is not saved when looking at the position with M851. When changing in the probeoffset then turning the power off then on again shows it did not store.

radek8, 2020-04-10:
The value is not saved when you click save. The save message is not displayed.

After commenting on the if condition, the save button works. The save message is displayed

PPacione, 2020-04-10:
After commenting out what you have above in the two files will it save the value entered in the probeoffset if I go to the I go to the Custom menu and press Save to EEPROM. During a print, going to the BabyStep menu, I need to press More first, if I make any changes and press SAVE I get a beep and a message telling me it saved. It does not save.

PPacione, 2020-04-11:
Your fix does correct the problem. This was my error. I did not enable BABYSTEP_ZPROBE_OFFSET in the printer boards firmware. Once enabled this works great.
Thank you

coreflake1, 2020-04-15:
Nice, you're fix solved it for me as well !

radek8, 2020-04-24:
It will be fixed in parseACK.c

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/608

closed 2020-04-24 by radek8