bigtreetech/BIGTREETECH-TouchScreenFirmware #556: [BUG] (short description)

radek8, 2020-04-09:
228/5000
Some extruder parameters do not display correctly - Steps per mm, Max Acceleration, Max Feed Rate.
When displayed in Marlin mode, the values are OK

My configuration
Bigtreetech TFT35 E V3.0.25.3
Bigtreetech SKR 1.4 - Marlin 2.0.5.2 - 2x extruder

[image]
[image]
[image]

radek8, 2020-04-11:
I found out where the problem is.
The FW displays do not load parameters if 2 Extruders are defined.
I suggest editing the ParseACK.c file.
please check and process.

Files:
[parseACK.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4465203/parseACK.zip)

preview edit

[image]
[image]
[image]

jimhiggs, 2020-04-11:
I have 3 extruders, it wont show 3 I see, although I do have all three set the same in marlin conf.

radek8, 2020-04-11:
However, it is possible to configure more extruders and hotends in Configuration.h. The firmware code is not adapted to this. Must be programmed
Before it is fixed, if you have 1 tool and 3 extruders, add additional lines to each item in the ParseACK.c file. As I do.

closed 2020-04-30 by radek8