bigtreetech/BIGTREETECH-TouchScreenFirmware #558: [BUG] (Stuck on Slect Mode, Touch mode Not working after a minute or so)

fenix8k, 2020-04-10:
### Description

I was adjusting my K values for Lin_Advance, it was all ok, after the printer finish, I turn it off (after waiting nozzle go under to 60C).
Then I turn it on Again, start checking some configuration from the Touch Screen and it goes out to the Choose mid section and there start the nightmare!, After that, I turn the printer Off/On several times, re-flashed the firmware like 20 times with different settings (my screen comes with 25.2 from factory), The firmware update go OKs, an interesting behavior is that if i leave the printer off for like 5 minutes and turn it on, the TouchMode Work OK for like a minute or 2, then it goes back to the selection screen.

I try turn it on Only with 5V and GND it don't change the behavior no mather the printer is connected or not, the Marlin mode Work Fine.

I try to reset the settings on the Screen and it's say it's Ok that i need to powercycle it, but it don't fix the problem, no idea what to do, i was going to burn the bootloader again but i can't find what pins i need to short to put the screen on bootloader load state

It's Posible that a Capacitor is Wrong or something?, it was working fine for line 1 week.

Thx!

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**
TouchScreen Mode Working

**Actual behavior**
Stuck on Selection Screen, some times Marlin mode Work

### Hardware Variant

SKR 1.4 Turbo
TFT35 E3 V3.0
TMC2130
Ender-5
### TFT Firmware Version & Main Board Firmware details

Marlin bugfix-2.0.x Last to Date 9/4/2020
Last TFT TouchScreen Firmware 25.3
### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4459490/Configuration.zip)

bigtreetech, 2020-04-10:
Hello, You mean that TFT will automatically jump to the mode selection menu in touch mode, and will not automatically jump in 12864 mode?

fenix8k, 2020-04-10:
Yes, I upload a video to show what happens, it automatically change to select menu, if I have luck, I can chose Marlin mode, if I choose TouchMode it try to change it and then it goes back to select menu, sometimes it get stuck at BIQU Logo when i change Mode to Touch.

It was working fine for a week, no idea what happen, I try Powering it Only with 5V and GND same behavior, if i leave it without power for like 5-10 mins and turn it on, it work OK for like a minute, i think the eeprom (on the screen) got corrupted or something like that or a faulty capacitor? (because it work ok for like a minute or so)

Plz Help! T.T

fenix8k, 2020-04-12:
So after trying allot of things i could figure out what is going wrong, to make it short, the problem is with the digitalizer and the metal case for the Screen on the Ender5, the scree permanently detect a touch, its the same as pressing the wheel constantly, so to solve the problem I just put some tape between the screen and the metal frame and problem solved, I think this is a fail in the design because the TFT Screen is a direct replacement for Ender3/5 Screen.

I think it's important to leave this info in some relevant place so people struggling with this could fix it really Easy

hfog, 2020-04-12:
Yes. This is a known problem. The case comes into contact with the screen and registers as a touch. Most of us solved this problem with washers used as spacers to keep the case from pressing on the screen.

fenix8k, 2020-04-12:
> Yes. This is a known problem. The case comes into contact with the screen and registers as a touch. Most of us solved this problem with washers used as spacers to keep the case from pressing on the screen.

If this is a know bug, it need to be specify some where, I use all my weekend to solve this issue, i try to find help on the web and nothing found.

Bigtreetech, put this on the Wiki with the complete format you need a SD card need to be in order to flash the firmware.

hfog, 2020-04-12:
Well, I'm not going to argue with you there. BTT hasn't got the best track record for documentation in general. But if you had done a search through the issues before posting yours, you probably would have found it.

hfog, 2020-04-12:
And just to be clear. I'm not criticising you for not finding it. I'm just letting you know that BTT is already aware of the issue. That's all.

fenix8k, 2020-04-12:
> Well, I'm not going to argue with you there. BTT hasn't got the best track record for documentation in general. But if you had done a search through the issues before posting yours, you probably would have found it.

You can post a link to this issue and its solution?

I really search for this xD

closed 2020-12-15 by stale[bot]