bigtreetech/BIGTREETECH-TouchScreenFirmware #561: [BUG] Print abourt not stopping heaters & not realese stepp. motors / USB TFT not work

DanielSan1000, 2020-04-10:
### Description

1. Print abourt not stopping heaters & not realese stepp. motors also 

2. It is not pissible to use USB on TFT35 for printing
Not possible to start printinf from usb. Exit from manu after click on gcode

### Steps to reproduce

1. During the print stop printing.
2. Trying to start printing from TFT USB

**Expected behavior**

**Actual behavior**

### Hardware Variant

Bigtreetech TFT35 & skr e3 mini v1.2 (Ender 3 setup)

### TFT Firmware Version & Main Board Firmware details

BIGTREETECH-TouchScreenFirmware V3.0.25.3

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

bigtreetech, 2020-04-11:
Hello, it should be fixed in #480 , please download the lastest version firmware and test it

DanielSan1000, 2020-04-11:
Ok. Thanks

closed 2020-04-11 by DanielSan1000