bigtreetech/BIGTREETECH-TouchScreenFirmware #568: [BUG] Can not change extruder number in Extrude menu

jimhiggs, 2020-04-11:
Just updated to .26 and now  can not change extruder number by pressing nozzle button. It is stuck at E0 I have 3 extruders.

jimhiggs, 2020-04-11:
I reverted to 25 and still can't change extruder. Strange I could before changing to 26.

bigtreetech, 2020-04-13:
Hello, Firmware defaults to only one extruder， you need to set to number of extruder to 3 in here
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/3e7503113e956dd99f71bfd00596541ed0281727/TFT/src/User/Configuration.h#L142

jimhiggs, 2020-04-13:
Thanks. i must have done that when I installed 25 back in November. Sorry for the time waste.

radek8, 2020-04-15:
Version 26 also does not correctly support multiple extruders
look at
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/556

closed 2020-10-06 by jimhiggs