bigtreetech/BIGTREETECH-TouchScreenFirmware #571: [BUG] (Firmware assumes heated bed)

pessimism, 2020-04-13:
### Description

Firmware assumes heated bed but has no visible option to disable it.  This leaves an icon on screen that I can't use and errors from marlin not understanding bed-specific commands such as M140

### Steps to reproduce

1. Use screen with SKR 1.3 or other board, with Marlin firmware, on a printer WITHOUT a heated bed

**Expected behavior**
Screen should auto disable heated bed specific code or provide a configuration.h flag to completely disable and remove heated bed support

**Actual behavior**
Errors after every print due to TFT firmware sending M140 command to a printer that can't understand it

### Hardware Variant
TFT35 v2.0
### TFT Firmware Version & Main Board Firmware details
Current for both

KCGreg, 2020-04-16:
I added the following feature request because I am having this issue as well.  This firmware would be much better if it didn't assume certain setups.  https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/574

closed 2020-11-03 by stale[bot]

oldman4U, 2021-01-15:
Hi.

Could you please check the latest version of the firmware. There you can define in config.ini line 208 "heated_bed:0 or 1. So it can be disabled in firmware, being able to disable icons depending on the configuration is already covered in another ticket.

Thank you