bigtreetech/BIGTREETECH-TouchScreenFirmware #577: TFT 35 V3.0 in LCD12864 simulator and Marlin 2.x

curtait, 2020-04-16:
How can make "LCD12864 Simulator" desappear from the screen.
So then make all graph and text, bigger.
Can change text from green to other color and background color too, will be a nice thing to do.

trevjonez, 2020-04-16:
> How can make "LCD12864 Simulator" desappear from the screen.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/Configuration.h#L70

> So then make all graph and text, bigger.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/Configuration.h#L78

> Can change text from green to other color and background color too, will be a nice thing to do.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/Configuration.h#L64

I found when changing the colors they were ignored and I had to go into the settings in the touch mode to change them. perhaps there is additional work that needs to be done to have these default values in configuration.h to actually take effect

trevjonez, 2020-04-17:
sounds like you may not have flashed the images with the firmware.

the color values in Configuration.h are ignored in .26 it seems. You need to go into the settings menu in touch mode and change them there.

curtait, 2020-04-17:
Yes.
Doing that way, work.
Thank you

ChrisXenon, 2020-05-02:
... so why leave the f*cking stuff in configuration.h? I just wasted an evening trying to figure this out. What a great way to annoy someone for very little effort. HOW HARD WOULD IT BE TO DELETE THE STUFF THAT WON'T WORK ANYWAY?

carl1961, 2020-05-02:
ChrisXenon, I would not answer you,

ChrisXenon, 2020-05-03:
Well, no one from BTT has ever answered me. The email address on trhe card in the box which brought my mini E3 was wrong. jesus. But I found the correct one and wrote to them there - nothing. And I've subm,itted several issueson on GitHub - NOTHING. SO yeah - I'm a little angry - that's justifiable in the face of consistently non-existant support and dimwitted software engineering.

carl1961, 2020-05-03:
Well 99% of the people here been doing all this code work for free. Tons of hours. You post your issues here properly, some one will see it.

ChrisXenon, 2020-05-03:
Carl, I don't understand. What code are people doing for free? I purchased my hardware from BTT and it came with software which I thought was written by BTT. Isn't that true? If not, then I have some apologising to do.

carl1961, 2020-05-03:
Original plain jane touch screen is BTT stock firmware that comes on the display.

All the extras ( unified live icons, bltouch, ect...) where done by other people that summited PR's to BTT to add to Touch Screen Firmware.
The display's have cam a long was in a short time because of these dedicated contributers who took time out of there life to do this, so there will be bugs to work out. No other touch screen I have has this kind of speedy progress and help. 

This is just a few of the many contributers who has got the BTT touch screen to where it it today.

@guruathwal 

https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware

Msq001

https://github.com/Msq001/BIGTREETECH-TouchScreenFirmware

@thisiskeithb

https://github.com/thisiskeithb/BIGTREETECH-TouchScreenFirmware

Acenotass

https://github.com/Acenotass/BIGTREETECH-TouchScreenFirmware

HappyCodingRobot

https://github.com/HappyCodingRobot/BIGTREETECH-TouchScreenFirmware

@luc-github

https://github.com/luc-github/BIGTREETECH-TouchScreenFirmware

thisiskeithb, 2020-05-04:
@ChrisXenon: This project is open source, so we donate time as life permits.

The full list of Contributors can be found [here](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/graphs/contributors).

As of this comment, there are [10 pull requests](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pulls) for bug fixes & improvements.

Things will be improved. Just hang tight.

carl1961, 2020-05-04:
Thanks thisiskeithb, that would have saved me a lot of typing lol

ChrisXenon, 2020-05-04:
Thanks for clarification. I knew the firmware was open-source, but that just means the manufacturer made their source available. What I didn't know is that many of the features have been added by private volunteers.

When we give money for hardware whose operation relies on the firmware, I think we have a reasonable expectation that it should work as advertised and if/when it doesn't - we have the right to complain. That's where I am. I have paid about £80 for the mini E3 and the TFT35 and my reasonable expectations have not been met in many regards. The TFT35 is very buggy. The documentation is poor which has wasted a lot of my time. Unless I re-wire things I can't have runout detection and the benefits of the TFT. That is certainly not made clear up-front, andf that is the fault of BTT, and that is who I was complaining to. I stand by that, and right now I'm considering returning the TFT. The whole situation is a stinking mess which I didn't want, didn't expect and didn't want to pay for.

Oh, and their support is ABSOLUTELY AND ENTIRELY ABSENT. They have not answered a single enquiry.

Quite seperately, let me thank the volunteers who are working without payment to make this work well as a system. You didn't get a penny of my £80 and you might well pull this off and end up with a superb system. If I were in your position I'd lament the fact that I am doing so much of the hard work whilst BTT employees are doing shoddy work and no support, and taking all the cash. But that's up to you..

When I complained here, I thought I was complaining to BTT staff. Are there any here? If not, then I should absolutely stop complaining here in the way I have done. But if they come here, then I'd quite like to keep smackling them aside the head for this fiasco. Please advise.

auroraonline001, 2020-07-21:
hi. new here. But i found that changing your required settings are available in the config.ini file rather than the configuration.h file works. as mentioned previously it seems to be a 26.1 'update'

ChrisXenon, 2020-07-21:
Thanks but even better -t hey are available in the user interface.

oldman4U, 2020-09-05:
@curtain

All the features you have been asking for (and many more), are now available in standard firmware.

Please close this ticket

Thank you

closed 2020-09-07 by curtait

ChrisXenon, 2020-09-07:
I've had two emasils askin gme to close this. It's already closed. What fun!

curtait, 2020-09-07:
Curtain & Curtait :)

ChrisXenon, 2020-09-07:
WHatever language you are using, I don't speak it.

curtain, 2020-09-07:
I'm not even a part of this repository or effort. Not sure how I even got included
s

---------------------
Scott Finkeldei
curtain@ksu.edu
Director of Academic and Student Technology
Kansas State University

AnHardt, 2020-09-07:
Amazing and amusing how much traffic a simple typo [here](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/577#issuecomment-687666893) can cause.
Please stop here.

oldman4U, 2020-09-07:
My fault. 

Sorry

oldman4U, 2020-09-07:
Not sure why i got curtain as a possible nickname.

Again I apologize and thank you for closing this ticket. 

@AnHardt Isn‘t it all about Being human and making mistakes 🤪

AnHardt, 2020-09-07:
Sorrrry could not rrrresist. https://www.youtube.com/watch?v=0hoa9tXnZX8

guruathwal, 2020-09-07:
@oldman4U  auto-correct on your iPhone did it for you... 😆

oldman4U, 2020-09-07:
@AnHardt Neue deutsche Welle! Das waren Zeiten, da war selbst ich noch jung and not an old man;-)))

@guruathwal How can you know that i have an iPhone;-) But in this case it was something else. I usually make an @ and then i get the members of this ticket. Somehow i got curtain by mistake... Happy it was not CURTcobAIN;-)))))

curtain, 2020-09-07:
Hey, I will take a Kraftwerk tune anytime. Best part of getting mis-tagged on this!

---------------------
Scott Finkeldei
curtain@ksu.edu
Director of Academic and Student Technology
Kansas State University