bigtreetech/BIGTREETECH-TouchScreenFirmware #578: [BUG] Icons missing from Classic UI/Theme

thisiskeithb, 2020-04-16:
### Description

There are several missing icons when updating using the Classic Menu UI/Theme since the menus appear to only be built for the Unified Menu:

- CoolDown.bmp
- global_bed.bmp
- global_nozzle.bmp
- Heat_Fan.bmp
- Home_Move.bmp
- InfoBox.bmp
- MainMenu.bmp
- manual_level.bmp
- printing_bed.bmp
- printing_fan.bmp
- printing_flow.bmp
- printing_nozzle.bmp
- printing_speed.bmp
- printing_timer.bmp
- printing_zlayer.bmp
- status_flow.bmp
- status_speed.bmp
- StatusBed.bmp
- StatusFan.bmp
- StatusNozzle.bmp

Reset.bmp wasn't listed as "not updated", but it's not in the Classic UI/Theme folder 

### Steps to reproduce

1. Grab pre-built binaries from this repo for Classic Menu/UI
2. Update TFT using an SD card
3. Watch for errors at the end listing the missing icons in the list above

**Expected behavior**
Icons for Classic UI/Theme

**Actual behavior**
Missing icons for Classic UI/Theme

### Hardware Variant
All

### TFT Firmware Version & Main Board Firmware details
V.26/most current as of today

Tagging @guruathwal since these icons should be updated when updates are made to the Unified Menu. Note: The Unified UI icons can be copied over, but they don't fit the Classic UI theme.

Please don't forget about the Classic UI!

guruathwal, 2020-04-18:
The global status icons were not added by me, the manual_level icons should have been added at the time of adding the manual levelling option.
Rest of the icons are not required in the Classic menu.
I do not know the source of the blue classic icons. They look like a mix and match whatever is in hands. Most of the icons do not even match the function they perform.
Maybe @kkaleta could provide a simple Classic Menu theme to replace the existing ones. He is the best at designing these icons.
I will push a fix to avoid updating icons that are not required.

guruathwal, 2020-04-18:
The new printing menu needs icons which are not available in the classic theme. They need to be copied from the Unified theme, which again makes it ugly with classic icons or else the classic icons also need to be maintained at the same pace of the developments.

Also, there will be updates in future which may make it hard to keep updating the classic menu option. Some of the effects can be seen in the Printing menu and Machine settings.

@bigtreetech are we sure to keep the classic menu option? this has not started to look like pulling the development backwards, same as making things compatible with Windows 98 while developing for Windows 10.

kkaleta, 2020-04-18:
@guruathwal, I fully agree. Maintaining the classical menu it's time wasting. This should be abandon long time ago.

Tillema, 2020-04-20:
The classic menu is great for drawing icon sets yourself. With the unified menu, you are more limited if you don't know how to program (e.g. shifting texts, changing font colors, etc).
Currently I'm working on a more modern minimalistic complete set, incl. the unified menu, but I can say that the programming is the hardest part.
For now, I made a few changes for myself: inverse Z direction, 3-point leveling and red/blue colored icons while heating/cooling (in the top bar).
I'm still working on it, but like the way how easy (well, not really easy, but doable) it is to change stuff yourself.

PR #583 (2020-04-25): Fix for missing icons in classic menu and minor bug fix for fan speed display
- Minor bug fix for fan speed display when SHOW_FAN_ PERCENTAGE is turned off.
- use 'reset_value' icon for resetting values to default in the baby-stepping menu and probe-offset menu.
- Fix for preventing status screen icons update in the classic menu firmware.

Resolves #578
closed 2020-04-25 by bigtreetech