bigtreetech/BIGTREETECH-TouchScreenFirmware #579: [BUG] Print confirmation issue on TFT35 3.0

mouse100, 2020-04-17:
Flashed the TFT35 E3 3.0 to latest official firmware but after I launch a print from SD/USB I have the print confirmation dialog behind the other TFT icons.

[image]

bigtreetech, 2020-04-17:
Hello， This display is indeed a bug. The root reason may be that you enabled the “Filament runout” feature but did not insert the “Filament runout sensor”. Please check your setting in "Settings->Feature->Filament runout”?

mouse100, 2020-04-19:
> Hello， This display is indeed a bug. The root reason may be that you enabled the “Filament runout” feature but did not insert the “Filament runout sensor”. Please check your setting in "Settings->Feature->Filament runout”?

@bigtreetech I have the feature disabled but the bug is still there.

[image]

rjacobs1969, 2020-04-19:
I've seen this on my TFT35 V3 as well (using 26 release) moreover when a popup  happens  in the file selection screen you can't see it nor dismiss (press ok) it. leaving you with no working touch until you reboot.
I've not had time to dive in the source code yet but it seems the pop-ups are painted with a lower priority (=below) than other screens but are blocking touch input until you press OK (which is not possible if other elements are drawn over the button)
Popups should be aways be drawn on top, also it would be nice if you can ok these by pressing the encoder.

oldman4U, 2020-09-07:
Hi.

What is the status of this issue?

Should be fixed in the meantime.

closed 2020-12-15 by stale[bot]