bigtreetech/BIGTREETECH-TouchScreenFirmware #582: [BUG] (short description) TFT3.( v3.0 : last firmware update no LCD12864 simulator

davidmarli, 2020-04-17:
Hi,

I flash the last firmware for tft 3.5 v3.0. (v0.26 of the bin file)
When flashing, it told me some bmp were missing.
The touchscreen function works well, but when I want to change for LCD12864, the screen remains black !!!
Before updating the firmware all was fine...

Thank you for your help

kerk1v, 2020-04-18:
Check in touch menu if LCD color is not set to black background and black letters.

davidmarli, 2020-04-18:
That was that !!!
Thank you very much.

All is working now..

closed 2020-07-04 by bigtreetech

oldman4U, 2020-07-05:
@bigtreetech 

Could you please also close the two there tickets from davidmarli as he seems to be good in opening new tickets but does not understand that they have to be closed once they are not needed anymore.

Thank you