bigtreetech/BIGTREETECH-TouchScreenFirmware #584: SKR E3 Mini + TFT35 E3 V3. babysteps doesn't work

nicolas81, 2020-04-18:
Baby steps doesn't work : When you click + or - on the touch screen, the nozzle doesn't move.

drphil3d, 2020-04-19:
I seem to be having this issue as well

xboxpro1, 2020-04-19:
Hello, have you uncomment the line
//#define BABYSTEPPING in Configuration_adv.h / Marlin Firmware?
<https://marlinfw.org/docs/configuration/configuration.html#babystepping>

nicolas81, 2020-04-19:
Yes of course. When I use Marlin mode, babystepping works well. I think that this issue come from TFT35 firmware.

xboxpro1, 2020-04-19:
I can confirm that babystepping is working while printing. TFT35 E3 V3 + SKR V1.4.

nicolas81, 2020-04-19:
I have SKR E3 MINI V1.2.
 BabySteps are working correctly with the Marlin screen but not with BTT TFT E3 V3 screen for me...

nicolas81, 2020-04-20:
I noticed today that babysteps are working correctly within the Classic Menu but not with in the Unified Menu.

xboxpro1, 2020-04-20:
I think the baby step commands are not sent because the tft rs232 buffer is full.
try to edit BabyStep.c / TFT Firmware. Replace the command "storeCmd" with "mustStoreCmd" in BabyStep.c. The command occurs 4 times in the file. And the compile and flash TFT firmware.

nicolas81, 2020-04-21:
I tried but I get this error :

[image]

nicolas81, 2020-04-21:
I understood that mustStoreCmd is a void instead of storeCmd is bool function. So I will try this modification
[image]

nicolas81, 2020-04-21:
Now, babysteps  are working  fine with + and - button. Thanks xboxpro1 !
Still remains problem with the encoder. As I am seeing in the code the encoder change the value of baby_step_value but doesn't send gcode command to the mother board. So the nozzle doesn't move with the encoder. And I don't understand the goal of the save button in this menu.

xboxpro1, 2020-04-21:
why dont you fork marlin fix the issue and make a pull request. you are right the save button is obsolete. you can remove the button. change {ICON_EEPROM_SAVE,          LABEL_EEPROM_SAVE} to
{ICON_BACKGROUND,           LABEL_BACKGROUND}, i have altered ProbeOffset.c in my fork, now the nozzle is moving when i set the z offset. the encoder works too, maybe you can have a look.

i use this altered function form Move.c

AXIS zAxis = Z_AXIS;

void storeMoveCmdZ(AXIS xyz, int8_t direction) {
  const char *xyzMoveCmd[] = {Z_GCODE};
  // if invert is true, 'direction' multiplied by -1
  storeCmd(xyzMoveCmd[xyz], (infoSettings.invert_axis[xyz] ? -direction : direction) * elementsUnit.ele[elementsUnit.cur]);
  // update now axis be selected
  zAxis = xyz;
}

And this code in the encoder loop

if(encoderPosition)
          {
           baby_step_value += elementsUnit.ele[elementsUnit.cur]*encoderPosition;
//this code   ---->         storeMoveCmdZ(Z_AXIS, encoderPosition > 0 ? 1 : -1);
            encoderPosition = 0;
          }

I'm just trying myself. i am not the creator or maintainer.

xboxpro1, 2020-04-21:
maybe you must use "mustStoreCmd" in the function, because the printer is busy.
when i level the z offset the print is idle and "storeCmd" is working fine.
void storeMoveCmdZ(AXIS xyz, int8_t direction) {
const char *xyzMoveCmd[] = {Z_GCODE};
// if invert is true, 'direction' multiplied by -1
mustStoreCmd(xyzMoveCmd[xyz], (infoSettings.invert_axis[xyz] ? -direction : direction) * elementsUnit.ele[elementsUnit.cur]);
// update now axis be selected
zAxis = xyz;
}

xboxpro1, 2020-04-21:
oh i forgot you have to define in babystep.c 
#define Z_GCODE "G1 Z%.1f\n"

nicolas81, 2020-04-21:
I think that I need to increase my skill in c++ programming before open a fork. I understand most of the code you give me above except %.2f. I guess it is a way to convert float into two digits float or something like that...but when I follow code with visual studio code I can't understand this.

xboxpro1, 2020-04-21:
stop, i made a big mistake , #define Z_GCODE "G1 Z%.2f\n" is not working!!

you have to define the babystep gcode..

#define Z_GCODE "M290 Z%.2f\n" <--- this could do it

sorry...

<https://www.codecademy.com/forum_questions/51382b87954cc276830008f9>

nicolas81, 2020-04-21:
Now, I know  what means %.2f. 
I managed to disable button "save" and my encoder control the nozzle too.
Thanks so much xboxpro1 !

nicolas81, 2020-04-21:
`#include "BabyStep.h"
#include "includes.h"

//1 title, ITEM_PER_PAGE items(icon+label)
MENUITEMS babyStepItems = {
//title
  LABEL_BABYSTEP,
//icon                        label
 {{ICON_DEC,                  LABEL_DEC},
  {ICON_BACKGROUND,           LABEL_BACKGROUND},
  {ICON_BACKGROUND,           LABEL_BACKGROUND},
  {ICON_INC,                  LABEL_INC},
  {ICON_BACKGROUND,           LABEL_BACKGROUND},   //{ICON_EEPROM_SAVE,          LABEL_EEPROM_SAVE},
  {ICON_01_MM,                LABEL_01_MM},
  {ICON_NORMAL_SPEED,         LABEL_VALUE_ZERO},
  {ICON_BACK,                 LABEL_BACK},}
};

typedef struct
{
  const ITEM *list;
  const float *ele;
  uint8_t cur;
  uint8_t totaled;
}ELEMENTS;

#define ITEM_BABYSTEP_UNIT_NUM 3
const ITEM itemBabyStepUnit[ITEM_BABYSTEP_UNIT_NUM] = {
// icon                       label
  {ICON_001_MM,               LABEL_001_MM},
  {ICON_01_MM,                LABEL_01_MM},
  {ICON_1_MM,                 LABEL_1_MM},
};
const float item_babystep_unit[ITEM_BABYSTEP_UNIT_NUM] = {0.01f, 0.1f, 1};

static ELEMENTS elementsUnit;

static void initElements(u8 position)
{
  elementsUnit.totaled = ITEM_BABYSTEP_UNIT_NUM;
  elementsUnit.list = itemBabyStepUnit;
  elementsUnit.ele  = item_babystep_unit;

  for(u8 i=0; i<elementsUnit.totaled; i++)
  {
    if(memcmp(&elementsUnit.list[i], &babyStepItems.items[position], sizeof(ITEM)) == 0)
    {
      elementsUnit.cur = i;
      break;
    }
  }
}

static float baby_step_value = 0.0f;

void babyStepReset(void)
{
  baby_step_value = 0.0f;
}

#define BABYSTEP_MAX_VALUE 5.0f
#define BABYSTEP_MIN_VALUE -5.0f

void showBabyStep(void)
{
  GUI_DispFloat(CENTER_X - 5*BYTE_WIDTH/2, CENTER_Y, baby_step_value, 3, 2, RIGHT);
}
void babyStepReDraw(void)
{
  GUI_DispFloat(CENTER_X - 5*BYTE_WIDTH/2, CENTER_Y, baby_step_value, 3, 2, RIGHT);
}

void menuBabyStep(void)
{
  KEY_VALUES key_num = KEY_IDLE;
  float now = baby_step_value;

  initElements(KEY_ICON_5);
  menuDrawPage(&babyStepItems);
  showBabyStep();

  #if LCD_ENCODER_SUPPORT
    encoderPosition = 0;
  #endif

  while(infoMenu.menu[infoMenu.cur] == menuBabyStep)
  {
    key_num = menuKeyGetValue();
    switch(key_num)
    {
      case KEY_ICON_0: // - button
        if(baby_step_value - elementsUnit.ele[elementsUnit.cur] > BABYSTEP_MIN_VALUE)
        {
          mustStoreCmd("M290 Z-%.2f\n",elementsUnit.ele[elementsUnit.cur]);
            baby_step_value -= elementsUnit.ele[elementsUnit.cur];
        }
        break;
      case KEY_ICON_3: // + button
        if(baby_step_value + elementsUnit.ele[elementsUnit.cur] < BABYSTEP_MAX_VALUE)
        {
          mustStoreCmd("M290 Z%.2f\n",elementsUnit.ele[elementsUnit.cur]);
            baby_step_value += elementsUnit.ele[elementsUnit.cur];
        }
        break;
      /*case KEY_ICON_4: // save 
        if(infoMachineSettings.EEPROM == 1){
           mustStoreCmd("M500\n");
        }
        break;*/
      case KEY_ICON_5: // change resolution
        elementsUnit.cur = (elementsUnit.cur + 1) % elementsUnit.totaled;
        babyStepItems.items[key_num] = elementsUnit.list[elementsUnit.cur];
        menuDrawItem(&babyStepItems.items[key_num], key_num);
        break;
      case KEY_ICON_6: // set to zero
        mustStoreCmd("M290 Z%.2f\n",-baby_step_value);
          baby_step_value = 0.0f;
        break;
      case KEY_ICON_7:  // exit button
        infoMenu.cur--;
        break;
      default :
        #if LCD_ENCODER_SUPPORT
          if(encoderPosition)
          {
            mustStoreCmd("M290 Z%.2f\n",elementsUnit.ele[elementsUnit.cur]*encoderPosition);
            baby_step_value += elementsUnit.ele[elementsUnit.cur]*encoderPosition;
            encoderPosition = 0;
          }
        #endif
        break;
      }
    if(now != baby_step_value)
    {
      now = baby_step_value;
      babyStepReDraw();
    }
    loopProcess();
  }
}
`

nicolas81, 2020-04-21:
For people who want to do the same thing as me, you have the entire code of BabyStep.c file above.

xboxpro1, 2020-04-22:
remember the  touch function "save" is still there, when you alter a icon with the placeholder you have to change the touch function too. when you touch on the placeholder the command will still be executed.
icon order:
0 , 1, 2, 3
4, 5, 6, 7

key_num = menuKeyGetValue();
switch(key_num)

you have to remove
case KEY_ICON_4:
        if(infoMachineSettings.EEPROM == 1){
           storeCmd("M500\n");
        }
break;

the tft touch mode is like a pc connected through the usb port, like the software pronterface.
you can excute all marlin gcode commands 
<https://marlinfw.org/docs/gcode/M500.html>

that concept is very simple but also very genius from bigtreetech.

now you have your own function you can use it for the touch icons too..

case KEY_ICON_0: //decrease
        if(baby_step_value - elementsUnit.ele[elementsUnit.cur] > BABYSTEP_MIN_VALUE)
        {
         // if(storeCmd("M290 Z-%.2f\n",elementsUnit.ele[elementsUnit.cur])) <--- remove
            baby_step_value -= elementsUnit.ele[elementsUnit.cur];
            storeMoveCmdZ(Z_AXIS, -1);     //<----new -1 for decrease
        }
        break;
      case KEY_ICON_3://increase
        if(baby_step_value + elementsUnit.ele[elementsUnit.cur] < BABYSTEP_MAX_VALUE)
        {
         // if(storeCmd("M290 Z%.2f\n",elementsUnit.ele[elementsUnit.cur])) <-- remove
            baby_step_value += elementsUnit.ele[elementsUnit.cur];
            storeMoveCmdZ(Z_AXIS, 1);     //<---new 1 for increase
        }
break;

take a look at Move.c

i suggest to make a fork and alter your file, you dont habe to make a pull request.
when btt release a new version you have to change your file again, so you can automatic merge
the changes .

or make a pull request , that a new release have your bugfix..

can you close the topic and mark it as solved?

have fun and happy coding....

nicolas81, 2020-04-22:
Thanks, I have ever remove the case "KEY_ICON_4"  in my code by using commented lines with /* and */ but when I copied and pasted code in GitHub window the * after and before the / have disappeared...now I think it's a GitHub bug. 😄

closed 2020-04-22 by nicolas81