bigtreetech/BIGTREETECH-TouchScreenFirmware #587: [BUG] tft35 v3 E3 disconnected to printer during bootup

CorruptAvocado, 2020-04-18:
on the tft35 v3 E3, if I boot up on the touch mode, the printer is disconnected. But if I go to the Marlin mode, the boot pictures load and I am connected to the printer. Can you make it so if I boot up on the touch mode, it will connect to the printer automatically?

Chronicmemecrafter, 2020-04-20:
The printer always boots up to touchscreen mode unless you alter the firmware. When the printer boots up, it takes time to connect just like before you had the tft35 except now it says not connected before it is ready. If it doesn't connect at all in touchscreen mode then that is because the touch cable (the 5 pin black one) isn't connected correctly or at all.

oldman4U, 2020-09-11:
Hi.

Any news on the reported issue?

Please close this ticket if you do not need it anymore. Otherwise let us know

closed 2020-12-15 by stale[bot]