bigtreetech/BIGTREETECH-TouchScreenFirmware #588: BTT SKR MIni E3 V1.1 isn't connecting with BTT TFT35-E3 V3.0

manu504, 2020-04-19:
Hello I recently bought a combo of Bigtreetech SKR Mini E3 V1.1 DIP and a Bigtreetech TFT35-E3 V3.0 i install the board in a CR-10 mini and it works perfect now i am trying to install the Touchscreen and i keep getting the errror No printer Attached i join Bigtreetech facebook group and did everythingh they wrote and still keep getting the same error i am running out of ideas and dont know if it is somethingh wrong configurate in the LCD section in marlin for my board or the TFT port in the Board doesn't work i checked wiring, connection and i think it is ok, I updated TFT35-E3V3.0 to .26 version and I dont have more ideas, could you please help me.

Chronicmemecrafter, 2020-04-20:
Link your config files and tell us how you have the screen connected in detail. Try not to be so vague saying "I think it is ok" I don't think this is a bug it is more of a personal issue that you probably have something wrong.

oldman4U, 2020-06-08:
Hello manu504.

Is this still an issue? Please let us know

Thank you

oldman4U, 2020-09-05:
@manu504 

Close this ticket.

oldman4U, 2020-09-11:
@bigtreetech 

Please close this ticket because the user does not reply anymore.

Thank you

closed 2020-12-27 by stale[bot]