bigtreetech/BIGTREETECH-TouchScreenFirmware #589: [Q] (can this get working with MKS TFT 35 v1.0?

rightawy, 2020-04-19:
i see this firmware working with MKS TFT32. I s this possible to work with MKS TFT35?  how can I get that printing screen?

thisiskeithb, 2020-04-19:
Of the MKS TFTs, only the MKS TFT32 1.4 is supported at this time.

rightawy, 2020-04-20:
cant you add it also?

thisiskeithb, 2020-04-20:
It can be done, yes. Someone will have to port code to the different hardware variants.

darkspr1te, 2020-06-18:
firmware for 3.5" screen (with stm32f407) is almost complete, am still awiaing my 3.5 screen to test the firmware on but it should work, i wrote a new bootloader for the 407 model without even having one. so am confident it will be usable

extremeshok, 2020-07-11:
if you can post it, i can test it for you

oldman4U, 2020-09-11:
Hello.

Any update on this topic?

darkspr1te, 2020-09-12:
For those wanting to try out this build, open source bootloader here https://github.com/darkspr1te/mkstft35_V1_open_bootloader/blob/master/buildroot/SDCARD/firmware.bin
binary of BTT for MKSTFT35V1_0 here https://github.com/darkspr1te/mkstft35_V1_open_bootloader/files/5210339/mkslcd.zip
fonts/bmp files for sdcard here 
https://github.com/darkspr1te/mkstft35_V1_open_bootloader/files/5210445/MKS.zip
and to revert to factory bootloader avilable here 
https://github.com/makerbase-mks/MKS-TFT/files/5156616/mkstft35_v1.0.3_bootloader.zip

oldman4U, 2020-10-21:
And now this is merged into master branch. Could you please test it and close the ticket afterwards. Thank you

closed 2020-10-23 by rightawy

chrisdaloa, 2020-11-01:
I can not see this options in platformio.ini, Is it possible to install it and try? Thanks

chrisdaloa, 2020-11-02:
how can I load the bootloader on this display? It's a mks-tft35 rebranded. There are no TX e RX pins..

[image]

darkspr1te, 2020-11-03:
can you confirm if this is a flying bear reborn board and lcd ? if so its using the stm32f103 cpu on the main board/tft35 lcd and the mkstft35 uses stm32f407+tft 35 which means the tft35/2.8/3.2 firmwares are not compatible (they uses stm32f107/407 )  , are there any links to manufacture pages/schematics ?