bigtreetech/BIGTREETECH-TouchScreenFirmware #591: [BUG] (short description)

Taomyn, 2020-04-20:
### Description

Screen brightness keeps dimming/brightening randomly both in BTT mode and Marlin mode, and the feature menu hangs going from 2nd to 3rd screen, have to switch off/on.

### Steps to reproduce

See description

### Hardware Variant
BIGTREE_TFT35_E3_V3_0
### TFT Firmware Version & Main Board Firmware details
Latest TFT source as at time of writing, also latest Marlin bugfix-2.0.x

### Additional Information
[files.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4505344/files.zip)

Chronicmemecrafter, 2020-04-20:
Okay well first of all #define DEFAULT_LCD_IDLE_TIMER is set to 7 when it should only be 1 for on and 0 for off. Also, try adjusting your baud rate down. What Ido when troubleshooting configs is to just compare them to the stock one's side by side and change things that aren't mandatory for the feature I want to add to see what is breaking.

Taomyn, 2020-04-21:
@Chronicmemecrafter 

[image]

My settings haven't changed for the last 3 builds I made, on 20200410, 20200411 and 20200417

Taomyn, 2020-04-25:
Just compiled the source from the release 4hrs ago, and apart from needing to rename 3 of the material icons that were missing "_" in their names, the feature menu is fixed. However the LCD brightness is still broken, even though you have now changed option 7 to be option 8, it still does not set the normal/dim brightness and timeout values correctly. Doing them in the menu on the machine does work.

closed 2020-08-14 by bigtreetech