bigtreetech/BIGTREETECH-TouchScreenFirmware #592: [BUG] TFT35 E3 V3.0 Knob LED control option not showing up after custom firmware flash.

Chronicmemecrafter, 2020-04-20:
So after I did a custom firmware flash with a cool dark icon theme I found on reddit, I noticed that in the Features menu, the option to change the lcd knob led color is no longer there. Haven't tried to print anything yet to see what that menu looks like but everything else looks normal.

Steps to reproduce

1. Compile firmware given below not in specified environment.
2. Flash firmware to lcd and have every bitmap load except percentage.bmp and something else that was to fast to read.

**Expected behavior**
Everything would load fine (as it did) and all options would be available

**Actual behavior**
Everything except knob led control is there

### Hardware Variant
I'm using the TFT35 E3 V3.0 with the SKR Mini E3 V1.2 on my Ender 3 Pro

### TFT Firmware Version & Main Board Firmware details
TFT firmware is basically stock except with the different bitmaps and different default changes to the background of marlin mode. Mainboard Firmware is full custom starting with BTT example files (linked below)

### Additional Information
Mainboard configs (custom Marlin 2.0.5.3): https://pastebin.com/uXdU6yaA (config.h) and https://pastebin.com/T4hL6fx3 (config_adv.h)
TFT configs (custom 3.0.26): https://pastebin.com/upckrLvL

Also I don't know if this is known or important at that but in order to flash the custom firmware I had to set the default_envs to the regular V3.0 and not the E3 one otherwise I would get Ilegal Flash App! upon flashing

thisiskeithb, 2020-04-21:
The only way to get LED knob control on the TFT35 E3 V3 is to use the `BIGTREE_TFT35_E3_V3_0` environment in `platformio.ini`.

Chronicmemecrafter, 2020-04-21:
> The only way to get LED knob control on the TFT35 E3 V3 is to use the `BIGTREE_TFT35_E3_V3_0` environment in `platformio.ini

Do you mean changing the default_envs or compiling in the E3 environment? Because I have tried compiling in the E3 environment with non E3 default_envs and it has given me The Illegal Flash App! error upon flashing.

thisiskeithb, 2020-04-21:
> The only way to get LED knob control on the TFT35 E3 V3 is to use the `BIGTREE_TFT35_E3_V3_0` environment in `platformio.ini`.

Change `default_envs = BIGTREE_TFT35_V3_0` to `default_envs = BIGTREE_TFT35_E3_V3_0` in `platformio.ini`

Chronicmemecrafter, 2020-04-21:
I tried this and once again (yes I re-compiled using the checkmark in the bottom left of vscode) I got the error Illegal Flash App!

thisiskeithb, 2020-04-21:
Have you tried the pre-compiled binaries from this repo?

Chronicmemecrafter, 2020-04-21:
> Have you tried the pre-compiled binaries from this repo?

I have and they worked, only I want to customize more things than just the bit maps so I need the non-compiled versions. Wish they had  a version for each tft of the files right before compiling so I could compare and see what’s wrong.

thisiskeithb, 2020-04-21:
I’ve flashed a TFT24, TFT35 V3 and TFT35 E3 V3 with firmware from this repo many times without issue (other than missing icons for Classic UI), so there’s probably something else going on.

Tried a different SD card?

You can also try changing one setting at a time to see when it breaks on your custom builds.

Chronicmemecrafter, 2020-04-21:
I don't think it is the sd card as It's the 128mb one that comes with the E3 mini and has worked for both board and TFT flashes in the past. It's only this flash that is causing this error. I'll put on classic firmware with altered icons for now while I test a new un-compiled build one setting at a time. I'll let you know if and when anything breaks.

Chronicmemecrafter, 2020-04-21:
Okay, that is very weird, so I just tried flashing the V3.0.26 E3 firmware and got the Illegal flash app error again (yes I tried it with and without the custom icons). But when I tried to flash the normal V3.0.26 it worked perfectly the and only feature being omitted is the led knob color. Tho one thing I need to work so that I can make an undertale meme!

Chronicmemecrafter, 2020-04-21:
Nvm, it was the sd card. I think when introducing the features from the E3 variant the sd card reached some sort of limit when flashing and thus caused the error. Ima close the thread cause I' just an idiot.

closed 2020-04-21 by Chronicmemecrafter

thisiskeithb, 2020-04-21:
Glad you got it worked out. Replacing the stock SD card is the first recommendation because they easily go bad.