bigtreetech/BIGTREETECH-TouchScreenFirmware #593: Filament runout not working

Neme77, 2020-04-21:
### Description

### Install a smart filament detector

### Hardware Variant

### BIGTREE_TFT35_V3.0.26  & Trigorilla 14 and Marlin 2.0.5.3 Bugfix

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

poumsoon, 2020-05-31:
I also have the same problem.
When printing through MKS MINI 12864, the filament sensor works well, but it does not work when connected to TFT28 or TFT35 V2.
Testing with the May 26 firmware has the same problem.

poumsoon, 2020-06-01:
> I also have the same problem.
> When printing through MKS MINI 12864, the filament sensor works well, but it does not work when connected to TFT28 or TFT35 V2.
> Testing with the May 26 firmware has the same problem.

I knew the cause of the problem. In the case of touch LCD, the filament sensor must be connected to the port of the LCD. Refer to the TFT LCD manual.
  I didn't expect anything at all, and I think I've been looking for this and that for three days.

closed 2020-11-03 by stale[bot]

oldman4U, 2020-12-21:
kisslorand recommends to do the following:
Send M75 to Marlin when print begins. Marlin doesn't know that the print is started so no filament runout warnings. It's not a bug.
You can put M75 in your Start-Gcode and you're set. That's what I did.

Please let us know if this solves your problem

Thank you