bigtreetech/BIGTREETECH-TouchScreenFirmware #594: Feature: Add button for Marlin Manual Mesh Bed Leveling

RidgebackZulu, 2020-04-21:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
In the original Marlin LCD display is the option for bed leveling using the 9 point mesh. This works extremely well on the Ender 3 Pro. I would like to have a button added for this feature please.

**Describe alternatives you've considered**
I can long hold on the LCD until it switches back to the original Marlin LCD, perform mesh bed leveling, then long hold to go back.

**Additional context**
None.

rebel1, 2020-04-27:
Hi,
you can test this branch https://github.com/rebel1/BIGTREETECH-TouchScreenFirmware/tree/MBL

RidgebackZulu, 2020-04-28:
Hi, I installed this, (BIGTREE_TFT35_V3.0_E3.26.bin and /TFT35) but do not see a button for mesh level. I still only see the four point manual. I tried both Unified Menu and traditional.

rebel1, 2020-04-28:
Hi, here is an precompiled binary for you.
[BIGTREE_TFT35_V3.0_E3.26.bin.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4543623/BIGTREE_TFT35_V3.0_E3.26.bin.zip)

RidgebackZulu, 2020-04-28:
Thank you! I see the MBL now! Onto testing!

RidgebackZulu, 2020-04-28:
Tested well! I ran a calibration cube, and the mesh leveling worked well! I appreciate your efforts on this and hope it gets mainlined soon. Great work!

sonny-c, 2020-06-15:
> Hi, here is a precompiled binary for you.
> [BIGTREE_TFT35_V3.0_E3.26.bin.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4543623/BIGTREE_TFT35_V3.0_E3.26.bin.zip)

I was going to create a new post asking about Manual Mesh Bed Leveling, but instead, I will follow up here.

Does this .bin file put Manual Mesh Bed Leveling in Touch mode?
If so, does it allow you to edit a specific mesh point, as you can in LCD mode?
I am currently using the Unified MenuMaterial theme.  If I flash this .bin file, do new icons need to be created for the MBL options?

charakaSamawry, 2020-08-06:
Hi rebel1, i use MANUAL_PROBE to do UBL would tht be a good branch for me to use??  (Also running skr mini v2 and tft34 e3 v3)

islonina, 2020-10-08:
> Hi, here is an precompiled binary for you.
> [BIGTREE_TFT35_V3.0_E3.26.bin.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4543623/BIGTREE_TFT35_V3.0_E3.26.bin.zip)

Hi

This doesn't work for me, i have half english half german menu and a lot of missing icon

oldman4U, 2020-10-20:
Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 

Thank you

closed 2020-10-22 by RidgebackZulu

oldman4U, 2020-11-02:
@rebel1 

Hi. Have you ever made a PR for this? If not, why not?

oldman4U, 2021-01-13:
digant73 added Mesh bed levelling, so I remove it from the FR list

PR #1188 (2020-10-28): Mesh Editor + improvements + bugfix (replace PR #1154)
**NEW FEATURES:**
* resolves #288
* resolves #594
* resolves #777
* Implemented #288 and #777 adding a Mesh Editor (under BL menu) and Mesh Tuner menus providing a visual mesh grid (gradient colors are configurable in config.ini) and a fine tune tool for mesh editing. The Mesh Editor supports:
1) MBL
2) UBL
3) ABL Bilinear

**BUG FIXES:**
* Fixed #1167 MBL process causing "Invalid mesh" error trying to load the meshes with M420 S1. MBL process is now based on use of standard move gcode
* Fixed #967 related to wrong Software Endstops enabli [... cut]