bigtreetech/BIGTREETECH-TouchScreenFirmware #595: [BUG] ABL option does not appear on boot Vx.x.26

frankt2012, 2020-04-21:
### Description

No ABL option on boot up. I need to hold down rotary knob to act like I am switching to Marlin but instead just go back to TFT35 and then ABL shows up. Also it will not save my z offset. Goes back to 0.00 every time.
### Steps to reproduce

1. Turn on printer.
2. Go to ABL.
3. No ABL listed.
4. Hold rotary knob to go to Marlin but just go back to Bigtree.
5. ABL then shows up

**Expected behavior**
ABL shows up in menu and z offset settings saved.

**Actual behavior**
No ABL in menu at all on boot up.

### Hardware Variant
BIGTREE_TFT35_E3_V3_0
### TFT Firmware Version & Main Board Firmware details
TFT Firmware - Vx.x.26
Marlin firmware - 2.02

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

Chronicmemecrafter, 2020-04-21:
I don't believe there is anything that should be affecting this from the TFT Firmware. Try updating to the newest version of Marlin as 2.0.2 is relatively old and you can find good, up-to-date firmware in a lot of places if you aren't that comfortable messing with firmware.

frankt2012, 2020-04-21:
I been sitting here messing with this thing I was thinking the same thing. It is an old version I am using apparently. I am using a version someone compiled that was already set up for my bltouch mount I am using. I guess I can copy the settings over. I just started messing with firmware pretty much yesterday when I tried editing the TFT35 firmware. I am gonna try tomorrow. Thank you.

Neme77, 2020-04-21:
I have the same bug with Marlin 2.0.5.3

frankt2012, 2020-04-21:
I am going to update Marlin but since

frankt2012, 2020-04-23:
Ok I am closing this because I was able to fix it by upgrading to Marlin 2.0.5.3. All is kinda good now but I can't get my z offset to save. Going to open another bug. Thank you.

closed 2020-04-23 by frankt2012