bigtreetech/BIGTREETECH-TouchScreenFirmware #596: [BUG] Screen Warning on bottom

el-quique, 2020-04-22:
#245 Description

I am seeing that when missing filament the warning of Filament "Runout" is at the bottom of the print screen.

[image]

### Steps to reproduce

1. No filament
2. Start Printing
3. Warning Filament Runout, see on the bottom
4. Clic on button "Confirm"

**Expected behavior**
Warning Filament Runout, see on the top

### Hardware Variant
MKS TFT 32/28
### TFT Firmware Version & Main Board Firmware details
TFT Firmware BigTreeTech xx.26
Marlin 2.0.4.4

### Additional Information
[Configuration.h.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4513133/Configuration.h.zip)

el-quique, 2020-04-27:
any idea

oldman4U, 2020-09-03:
Hi.

Is this still a problem, please let us know?

If not, please close this ticket

Thank you

closed 2021-01-04 by stale[bot]