bigtreetech/BIGTREETECH-TouchScreenFirmware #597: ESP32 CAM Compatibiliti

AntoszHUN, 2020-04-22:
Hi!

Did I have an idea that could solve the ESP32 camera compatibility?

I thought of this camera: https://www.aliexpress.com/item/32989244490.html?spm=a2g0o.productlist.0.0.5a845cdcd5Sc5o&algo_pvid=3460a8ab-40c9-4828-a85a-17ae83c0d304&algo_ -0 & btsid = 0ab6f82315875324915706456e4250 & ws_ab_test = searchweb0_0, searchweb201602_, searchweb201603_

TFT displays communicate on a UART port. The camera knows that too. There is also WiFi in it. You could also view the printer remotely while printing. Or save to memory card. What are your opinions? Because it has WiFi in it, there would be no need for the ESP-01S WiFi module.

I would be interested in opinions and opportunities!

Thanks!

Hello, AntoszHUN

oldman4U, 2020-10-20:
And this also...;-)

Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 

Thank you

AntoszHUN, 2020-10-21:
Ok. Thanks.

closed 2020-10-21 by AntoszHUN