bigtreetech/BIGTREETECH-TouchScreenFirmware #598: [FR] Implement "lvgl"

GeckoSys3D, 2020-04-22:
It seems wonderful GUI's are made with lvgl (https://github.com/littlevgl/lvgl).
MKS implemented this now. It would be awesome, if someone is able to implement this GUI library.

oldman4U, 2020-09-07:
Is this somehow related?

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1067

oldman4U, 2020-10-20:
Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 

Thank you

closed 2020-10-22 by GeckoSys3D

Domush, 2021-03-01:
Why are unfinished feature requests being closed?

Having a consolidated list doesn't fulfill a feature request, it only makes them easier to find.

This needs to be reopened, as do ALL of the feature requests which have not been fulfilled.