bigtreetech/BIGTREETECH-TouchScreenFirmware #603: [BUG] Z offset not saving

frankt2012, 2020-04-23:
### Description

When going into ABL menu and setting z offset and babbysteps it will not save. I lose the settings once the printer is turned off and back on. 

### Steps to reproduce

1. Go to ABL menu
2. Z offset and set it.
3. Hit save.
4. Turn off printer and then back on

**Expected behavior**
Z offset setting should say 1.10 which I set it to.

**Actual behavior**
Z offset says zero on fresh start up.

### Hardware Variant
TFT35

### TFT Firmware Version & Main Board Firmware details
TFT firmware is x.x.26
Marlin is 2.0.5.3

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

junkerpat, 2020-04-23:
Same for me !!!!!!!

roomme13, 2020-04-23:
I believe that save button not call M500 for EEPROM  save.

I don't know it bug or not, same behavior if you using 128x64 lcd.

It would be nice if it can save automatically to EEPROM.

Maybe I would like to add this. 

-For baby stepping, even you configure babystepping_probe in marlin. It show independent value in TFT. 
-If you edit z offset value in machine parameter, it cannot be minus at first, you need to enter any value  then delet, after that you can have minus value.

radek8, 2020-04-23:
I have the same problem.
The problem is in the file
Menu \ ProbeOffset.c
and
Menu \ BabyStep.c
comment out these dba lines and save will work

viz:
 https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/554

[image]

roomme13, 2020-04-23:
> 
> 
> I have the same problem.
> The problem is in the file
> Menu \ ProbeOffset.c
> and
> Menu \ BabyStep.c
> comment out these dba lines and save will work
> 
> viz:
> #554
> 
> [image]

if you clone latest, this is already commentout. but i cannot find parser for that EEPROM status.

you can put this on  parseACK.c and it will work.
[image]

junkerpat, 2020-04-23:
Working thanks

frankt2012, 2020-04-24:
> if you clone latest, this is already commentout. but i cannot find parser for that EEPROM status.
> 
> you can put this on parseACK.c and it will work.
> [image]

I have to try that. I did just recently find out that if I set it and go to Save EEPROM it then saves.

radek8, 2020-04-24:
The addition of passeACK.c works
Thank you

closed 2020-08-19 by bigtreetech