bigtreetech/BIGTREETECH-TouchScreenFirmware #604: [BUG] (short description)

ManiacCreality, 2020-04-23:
Doesn't show temperature while heating up. But works when print starts.

skaman82, 2020-04-24:
can confirm this as well

bigtreetech, 2020-04-25:
hello? Which source are you printing from? TFTSD or OnBoardSD?  it will cause this issue

> Doesn't show temperature while heating up

 if you print from OnBoardSD,  Because of the synchronization problem of Marlin(M109/M190)

ManiacCreality, 2020-04-25:
From OnBoardSD. Ok so printing from TFTSD will fix it? Will this be fixed by Marlin in future?

bigtreetech, 2020-04-26:
There is no such problem when printing from TFTSD. In fact, this is the feature of M109 / M190 not a Marlin bug.   We are planning to improve the functions of the OnBoardSd part to see if there is any perfect solution

cfx69, 2020-05-16:
Per this reddit: https://www.reddit.com/r/BIGTREETECH/comments/fn2zep/hard_to_summarize_problem_tft_35_e3_skr_mini_e3/     Is the problem not serial port issue? Seems to be resolved by user Oburi85 

17 days ago

Still got the problem with the refreshing display? Got the same thing too and got it working! Screen refreshes while printing from onboard SD Card.

I had to switch my Serial Ports in Marlin. This is how it looks like now:

/** * Select the serial port on the board to use for communication with the host. * This allows the connection of wireless adapters (for instance) to non-default port pins. * Serial port -1 is the USB emulated serial port, if available. * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader. * * :[-1, 0, 1, 2, 3, 4, 5, 6, 7] */#define SERIAL_PORT -1/** * Select a secondary serial port on the board to use for communication with the host. * :[-1, 0, 1, 2, 3, 4, 5, 6, 7] */#define SERIAL_PORT_2 0

Hope this fixes your problem too. :)

oldman4U, 2020-09-03:
@ManiacCreality 

This problem is solved i the meantime.

Please update to the latest firmware and close this ticket after that

Thank you

closed 2020-09-25 by ManiacCreality

oldman4U, 2020-09-25:
👍🏻