bigtreetech/BIGTREETECH-TouchScreenFirmware #610: [BUG] (Pause/Resume is broken)

pessimism, 2020-04-24:
### Description
Pause/Resume print functionality is broken.  Pressing pause stops the print but pressing resume does nothing, and after a time the option to resume disappears

### Steps to reproduce

1. Start a print
2. Press pause.  The print stops.
3. Press resume.  Nothing happens.

**Expected behavior**
Print resumes

**Actual behavior**
Nothing

### Hardware Variant
TFT35 v2.0
### TFT Firmware Version & Main Board Firmware details
TFT firmware v26.  SKR 1.3 with Marlin 2.0.5.3

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

bigtreetech, 2020-04-25:
Hello， Which source are you printing from?  TFT SD or OnBoardSD?

pessimism, 2020-04-25:
Hello, I am printing from the onboard SD

bigtreetech, 2020-04-27:
if printing from onboard SD, TFT will pause/resume printing by send "M25/M24", Please use the PC software such as printrun to check whether it can use "M25 / M24" to “pause / resume" Marlin normally

pessimism, 2020-04-27:
Printing via USB with Cura 4.6, pause/resume works

redonebe, 2020-05-14:
I have the same problem but with a MKS TFT. If you press resume the print head moves back to where it left off but nothing happens. If i wait long enough printing resumes. I am printing from SD. The same thing when a filament break is detected. After pressing resume I have to wait 8 minutes before the printer continues.

closed 2020-11-03 by stale[bot]