bigtreetech/BIGTREETECH-TouchScreenFirmware #612: [BUG] (Can't revert to classic UI after installing Unified UI)

pessimism, 2020-04-24:
### Description

Can't revert to classic UI after installing Unified UI

### Steps to reproduce

1. Install firmware with classic UI
2. Install firmware with Unified UI
3. Install firmware with classic UI

**Expected behavior**
UI reverts to classic

**Actual behavior**
firmware upload complains about missing graphics then displays Unified UI with a jumble of icons from classic and unified UIs

### Hardware Variant
TFT35 v2.0
### TFT Firmware Version & Main Board Firmware details
TFT v 26, Marlin 2.0.5.3

### Additional Information
[Configuration Files](https://github.com/pessimism/TronxyX1/tree/master/Marlin%202.0%20Config/SKR%201.3%20and%20TFT3.5%20v2.0)

Joeskaife, 2020-04-27:
I've noticed that some of the filenames in the release bmp directories dont line up with the source code.

There's a file icon_list.inc that has the names (buried in X macros).

I know from my own use that all the files starting "status" should start "status_", there are probably more.

Make a note of any missing .bmp files (I used my phone to video the startup sequence!) and have a look in the bmp directory to see if there;s something with a similar-but-not-exactly-right name and rename the bmp file!

sulieman-X, 2020-05-09:
I have the same problem what is the solution

Joeskaife, 2020-05-09:
> I have the same problem what is the solution

They have released a fix for this - but broken other things!

I suggest that you just go into the bmp directory (eg TFT35/bmp) and rename the files:

statusBed.bmp -> status_bed.bmp
statusFan.bmp -> status_fan.bmp
statusNozzle.bmp -> status_nozzle.bmp

There are some other new problems that may look like this: they have introduced new "%" icons which you may not have and because of the rather disappointing way that the XMacros work for icon names, being out of step means you just get garbage.

It'd be great if when someone comes up with a wonderful new icon, they stuck it ON THE END so the enumerated list doesn't get rebuilt for ALL the icons.

closed 2020-07-04 by bigtreetech