bigtreetech/BIGTREETECH-TouchScreenFirmware #614: Print From OnboardSD "Echo Unknown Command M33" then crash [BUG]

BritNoob, 2020-04-24:
When going into the Print Menu then selecting OnboardSD using the Unified BIGTREE_TFT35_V3.0_E3.26.bin firmware + TFT35 directory.
Print From OnboardSD will cause a..... 
"Echo , Unknown Command M33 *filename.gcode* " then crash and the display will become unresponsive. 

I have included a zip with the gcode. From looking at it i cannot see any M33...
Thanks!

[CFFFP_Test 2.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4531411/CFFFP_Test.2.zip)

thisiskeithb, 2020-04-25:
You need to enable `LONG_FILENAME_HOST_SUPPORT` in Marlin.

fibaq, 2020-04-25:
Hi I have same problem I ‘am new of this
Where I can find  
Please point me direction (folder) 
Thank you

bigtreetech, 2020-04-25:
https://github.com/MarlinFirmware/Marlin/blob/0518dec60d0931745efa2812fa388f33d68cfa29/Marlin/Configuration_adv.h#L1122

you need to enable `LONG_FILENAME_HOST_SUPPORT` in your Marlin `Configuration_adv.h` file

fibaq, 2020-04-25:
Hi Thank u for information but unfortunately 
Same error 
After enabled  This happened after update software to screen 
Any advice ? 
[image]
[image]

thisiskeithb, 2020-04-26:
Did you actually update your motherboard firmware? Marlin will return "Unknown command" when that particular G/M code(`M33` in this case) is not enabled.

xoniq, 2020-05-06:
> You need to enable `LONG_FILENAME_HOST_SUPPORT` in Marlin.

I got this issue too. This helped. Turned it on, compiled, and now it's working properly.

fibaq, 2020-05-07:
Hi yes working now but not printing from on board Sd

Leurie02, 2020-07-04:
> You need to enable `LONG_FILENAME_HOST_SUPPORT` in Marlin.

Hello... just don't use _ or any special characters on your file name :)

Leurie02, 2020-07-04:
> When going into the Print Menu then selecting OnboardSD using the Unified BIGTREE_TFT35_V3.0_E3.26.bin firmware + TFT35 directory.
> Print From OnboardSD will cause a.....
> "Echo , Unknown Command M33 [...]

Hello... just don't use _ or any special characters on your file name :)

closed 2020-08-19 by bigtreetech