bigtreetech/BIGTREETECH-TouchScreenFirmware #616: Display update error

AntoszHUN, 2020-04-25:
I want to signal a very annoying bug. When I start printing a file. I click on the file. Then, after the print screen appears, the hotend and hotbed temperatures are not shown on the display. Specifically, it does not update while heating the hotend or hotbed. Is there anything you can do about it?

Grrby, 2020-04-26:
I can confirm. I observe exactly the same bug and nothing has changed since several firmware versions. Temperatures are not updated until printing starts.

Another thing is huge lag in reaction. For example from touching Stop to real stop printing its about 20s. Stop button is mostly for some emergency situation and now is useless. But this is another topic.

AntoszHUN, 2020-04-26:
Then we are waiting for the repair. Or see if we get info from someone. :)

guruathwal, 2020-04-27:
This is not a bug. This is a Marlin Firmware issue because of **M109** & **M190** G-codes.
The **Marlin Firmware** gets stuck in a wait loop with these commands and waits for the temperature to reach the target. While in the wait loop it will not respond to any command until the temperature reaches the target or by sending **M108** to break the loop, which needs `EMERGENCY_PARSER` to be enabled in **Marlin Firmware**.
Because of this, the TFT will not receive any new temperature info and it will keep showing the last temperatures it received before the wait loop.

https://marlinfw.org/docs/gcode/M109.html

[image]

Grrby, 2020-04-27:
Thank you very much for explanation.

AntoszHUN, 2020-04-28:
Thanks for the reply! So does that mean you can't do that? I have EMERGENCY_PARSER turned on. Or should I turn it off? What's better?

guruathwal, 2020-04-28:
@AntoszHUN keep it turned on.

AntoszHUN, 2020-04-28:
That's right. Thanks!

closed 2020-04-28 by AntoszHUN