bigtreetech/BIGTREETECH-TouchScreenFirmware #617: [BUG] MKS-TFT32 not booting after flashing firmware

aero4, 2020-04-26:
### Description

When I copy the MKS folder and the MKS firmware (name changed to mkstft28.bin) from the directory "Copy to SD Card root directory to update" to my SD card the FW is being flashed, but stucks at 100% and manual rebooting ends with the display only showing "booting"...

### Steps to reproduce

1. Copy respective repository content (MKS folder and firmware .bin file) to SD card
2. Rename firmware to mkstft28.bin
3. Power cycle MKS TFT32

**Expected behavior**
Firmware is flashed and boots

**Actual behavior**
Firmware s flashed but not booting

### Hardware Variant
MKS TFT32_L V3.0
### TFT Firmware Version & Main Board Firmware details
MKS TFT28-32 Firmware V3.0.5

I tried that with a 16GB WIFI-SD card and a 128MB SD card

thisiskeithb, 2020-04-26:
> Hardware Variant
> MKS TFT32_L V3.0

Of the MKS TFT’s, only the TFT32 1.4 is supported.

aero4, 2020-04-26:
Only the hardware version 1.4? Isn't this version deprecated? What has to be done to get it workng on my TFT?

thisiskeithb, 2020-04-26:
> Only the hardware version 1.4?

Correct. Someone will have to port code to get it to work on other variants.

aero4, 2020-04-26:
Ok I understand...

closed 2020-04-26 by aero4

karabas2011, 2020-06-17:
MKS TFT32_L v4.0 works.

weed2all, 2020-09-20:
i have an mks tft 32 v3.0, does this work?