bigtreetech/BIGTREETECH-TouchScreenFirmware #620: Can only print from TFT SD card.

Basher1896, 2020-04-26:
### Description "A"

<Cannot print directly from the "U" drive>

### Steps to reproduce

1. [Insert 16gb flash drive]
2. [Select print]
3. [Select file to print]

**Expected behavior**
<Heating of nozzle and bed to initiate>

**Actual behavior**
<Tries to immediately print in midair with a warning about cold extrusion>

### Description "B"

<Cannot print from onboard SD>

### Steps to reproduce

1. [Insert 16gb SD card]
2. [Select print]
3. [Select file to print]

**Expected behavior**
<Heating of bed and nozzle to initiate>

**Actual behavior**
<No response>

### Hardware Variant
<BIGTREE_TFT35_V3.0_E3>
<BIGTREE_SKR_Mini_V1.2>
<BIGTREE_TFT35_V3.0_E3.26, Marlin-2.0.0.4.4>

### Additional Information

[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4535365/Configuration.zip)
[Configuration_adv.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4535366/Configuration_adv.zip)

bigtreetech, 2020-04-27:
Hello, Do you have a U disk formatted as a FAT32 file system?  Can't you read the U disk at all or can read the U disk file, but can't start printing?

closed 2020-04-27 by Basher1896

Basher1896, 2020-04-27:
Hello
Sorry I have just created a new issue (625)as there was information missing from my original upload.
Thank You
From: BigTreeTech 
Sent: Monday, April 27, 2020 4:28 AM
To: bigtreetech/BIGTREETECH-TouchScreenFirmware 
Cc: Basher1896 ; Author 
Subject: Re: [bigtreetech/BIGTREETECH-TouchScreenFirmware] Can only print from TFT SD card. (#620)

Hello, Do you have a U disk formatted as a FAT32 file system? Can't you read the U disk at all or can read the U disk file, but can't start printing?