bigtreetech/BIGTREETECH-TouchScreenFirmware #624: Update or not, that's the question it seems.

oldman4U, 2020-04-27:
I have an Ender3 Pro which works fine with a SKR E3 DIP, 2208 drivers and the standard TFT display. There is also a TFT35 E3 waiting to be installed.

Based on the many Bug reports related to the latest firmware, i ask myself if it is a good idea to change the display now, to wait until a more stable version becomes available, or maybe to use an older version, which has less features but which is more stable.

I believe we all agree when i say that new features are always welcome but they are useless in case they are full of bugs and even worse - brake other things which have been working fine before.

Your advice is highly appreciated.

Thank you

oldman4U, 2020-05-07:
After spending some days with the TFT35 testing different versions of the firmware i can say that a touchscreen in general is much better than an LCD. The two BTT touchscreens i have work great, but it would be great to have a comparison chart describing the difference in functions, because size is just one factor. And there are some well documented hardware related differences, but also some not so well documented functions like the ability to dimm the display.

The firmware is still missing many functions and has many bugs, but in general firmware version  x.x.26.1 seems to work well with the SKR Pro and the E3 DIP i use.

Conclusion for me without knowing many other solutions.

If you are willing to upgrade your printer from an LCD to a touchscreen - go for it. Based on the low price you have to pay for it, you have to spend a lot of time to make it work.

My personal favourite is the TFT35. Not because of the size but because of the speed and the other functions it offers for a little more. USB disk support and WIFI connection just to name two.

ssombra, 2020-05-07:
In my opinion it also depends on your workflow. I usually print using Octoprint and, as far as I know,  with some (most?) touchscreens you don't get current print information in the screen unless you send the print from the touchscreen itself. There are some workarounds like connecting the computer with octoprint (a Raspberry Pi in my case) directly to the touchscreen and not to the board, but then you have other issues like blobs (not fast enough). I decided to keep the touchscreen to control some things in the printer and because I like its functionality and look, but I also have a LCD connected.

oldman4U, 2020-05-08:
Hello ssombra.

Thank you for your post. Octoprint is great for sure, but most users including me struggle to get a working solution using hardware from BTT and Marlin FW. Functionality wise it also offers 99% of the things you really need for a working solution. The problems are missing or bad communication, (outdated) documentation and way too much bugs in the FW. As a beginner it takes much too long to get the things done, especially when you really expect drag and drop replacement of existing parts - like advertised. I am willing to spend hours to find things out and to learn new things, but that's me and i can imagine that others are not like this and therefore give up before they reach their goal. Look at the many open tickets here, where the person who initiated the ticket simply does not answer any more. I believe that many of them gave up, where some do not know how to close it and some do not take care - for sure.

Happy printing;-)

ssombra, 2020-05-08:
I agree with you, these solutions are not for everyone and they are definitely not drag and drop replacements. I also enjoy spending time making things work, for me it's a learning experience, and I have found a lot of support in forums, Facebook groups and here also. 
Take care

oldman4U, 2020-05-13:
I close this ticket, because there are already too many tickets which makes it hard to find the really important and actively used tickets.

closed 2020-05-13 by oldman4U