bigtreetech/BIGTREETECH-TouchScreenFirmware #625: Can only print from TFT SD card.

Basher1896, 2020-04-27:
### Description

Cannot print directly from the "U" drive

3 Steps to reproduce

1. Insert 16gb Fat32 flash drive.
2. Select print.
3. Select file to print.

Expected behavior :-
Initiate heating of Nozzle and Bed.
Actual behavior:-
Commences printing in mid air whilst warning about cold extrusion.

Hardware Variant
BIGTREE_TFT35_V3.0_E3
BIGTREE_TFT35_V3.0_E3.26
BIGTREE_SKR_MINI_v1.2
MARLIN-2.0.0.4.4

Additional Information
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4539198/Configuration.zip)
[Configuration_adv.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4539201/Configuration_adv.zip)

fftkatarn, 2020-04-30:
Similar behaviour here, no problem if printing from SD card of the screen but when I try to print from the motherboard slot the printer freezes on the print screen with the nozzle/bed temperature set to 0.

SulzRohr, 2020-06-18:
I have the same problem as Basher1896. Is there already a solution to this?!?

sbstnzmr, 2020-06-18:
I thought it was fixed. Last time I compiled the firmware it worked. Not sure if this is already within the pre compiled binaries.

oldman4U, 2020-09-03:
Hi.

What is the status of the reported issue?

Please close the ticket if you do not need it anymore

Thank you

oldman4U, 2020-09-11:
@Basher1896 

Close this ticket or reply.

closed 2020-09-12 by Basher1896