bigtreetech/BIGTREETECH-TouchScreenFirmware #628: [FR] Diasble popups or move to info box

Grievousbee, 2020-04-27:
First off apologies if this is not correct format but I dont felle the need to post here often.

Is it possible to add either a menu command or something in the cofiguration.h file to disable the popup messages.

I have an SKR1.4 and the 3.5 V3 touchscreen on my printer, a very heavily modified FLsun cube, and I am running a dual extruder set up. Chimera as it happens.

When ever a hot end become active i get a "Extruder n active" pop up that covers other displays, yes I can press ok but why?. This message is not needed as a pop up.
I would like this info to either be suppresible or put into the info box where it might actually be useful.
The same also goes for the "fade height disabled" or "Fade height = nn mm" pop-up.

I know what the fade height is I set it I don't need a pop-up telling me.

Pop-up should really ONLY be  used for things tyhat NEED confirmation.

lqbombjack, 2020-04-29:
Edit the following file: TFT/src/User/API/parseACK.c

Near the top you'll see a char array:
// Ignore reply "echo:" message (don't display in popup menu)
const char *const ignoreEcho[] = {
  "busy: processing",
  "Now fresh file:",
  "Probe Z Offset:",
  "Flow:",
  "echo:;",
  "echo:  G",
  "echo:  M",
};

Add:
  "echo:Active Extruder: 0",
  "echo:Active Extruder: 1",
  "echo:Active Extruder: 2",

Mine looks like so:
// Ignore reply "echo:" message (don't display in popup menu)
const char *const ignoreEcho[] = {
  "busy: processing",
  "Now fresh file:",
  "Probe Z Offset:",
  "Flow:",
  "echo:;",
  "echo:  G",
  "echo:  M",
  "echo:Active Extruder: 0",
  "echo:Active Extruder: 1",
  "echo:Active Extruder: 2",
};

Compile and copy to an SD Card, place it in the TFT board and reset the board to flash the new firmware.

lqbombjack, 2020-04-29:
You'll need to adjust to the number of extruders you have.  You can probably also do the same for the fade height if it's an item being echoed out by Marlin over the serial connection.

Grievousbee, 2020-05-01:
Ok have got most of the annoying stuff off now but fade height is giving a problem.

Do I have to put something like "Fade Height = xx mm" where xx is a height which would mean a string for every height or is there someway of saying all fade heights.

Also can the "Emergancy parser disabled" pop up on the Ender 3 be disabled in the same way.

Not sure why we even get this as it seems the emergancy parser bit os not enabled on the BTT SKR E3 Mini yet.

Thanks for the help.

thisiskeithb, 2020-05-01:
> Also can the "Emergancy parser disabled" pop up on the Ender 3 be disabled in the same way.

> Not sure why we even get this as it seems the emergancy parser bit is not enabled on the BTT SKR E3 Mini yet.

`EMERGENCY_PARSER` is not available on `STM32`-based boards which is what makes up most of BigTreeTech’s motherboards.

I agree that it’s not helpful for most users since they can’t enable it, even if they wanted to.

Grievousbee, 2020-05-01:
Seems to be ok on the SKR 1.4 though as I get no problems when I compile for that board with emergency parser enabled whereas it wont even compile on the E3 mini.

thisiskeithb, 2020-05-01:
> Seems to be ok on the SKR 1.4 though as I get no problems when I compile for that board with emergency parser enabled whereas it wont even compile on the E3 mini.

That’s because the SKR 1.1 (non-pro), 1.3, and 1.4 are `LPC1768/9`-based, which `EMERGENCY_PARSER` has been implemented on.

It won’t work on any other BigTreeTech board.

Ompapamowmow, 2020-05-05:
> Ok have got most of the annoying stuff off now but fade height is giving a problem.
> 
> Do I have to put something like "Fade Height = xx mm" where xx is a height which would mean a string for every height or is there someway of saying all fade heights?

I'm having the same problem. At the start of every print I was getting the "Fade Height Off" pop up so I put "M420 S1 Z10.00" into my start Gcode to set fade height to 10mm and now I get a pop up that says "Fade Height = 10mm". Can I enter something here TFT/src/User/API/parseACK.c to get rid of it?

dp250f, 2020-05-31:
I've added the following 2 lines to my **parseACK.c** file to **hide echo:Fade Height** and **echo:Bed Leveling** messages when **M420** is called in my Start G-Code:

// Ignore reply messages which starts with following text (don't display in popup menu)
const char *const ignoreEcho[] = {
  "busy: processing",
  "Now fresh file:",
  "Now doing file:,"
  "Probe Offset",
  "Flow:",
  "echo:;",     //M503
  "echo:  G",   //M503
  "echo:  M",   //M503
  "Cap:",       //M115
  "Config:",    //M360
  **"echo:Fade",    //M420
  "echo:Bed",    //M420**
};

In my opinion, these lines should also be in there by default.
"echo:Fade", //M420
"echo:Bed", //M420

DrumClock, 2020-07-26:
@lqbombjack 
I added
"echo: Active Extruder: 0",
"echo: Active Extruder: 1",

But the "echo: Active Extruder: 0" window still pops up when changed to T0.
This is OK when changing the extruder to T1.

lqbombjack, 2020-07-26:
I note in your quoted text you appear to have a space character between the colon after echo and the capital A of the word active.

Looking back at the strings of text I added there was no space between those characters. Could it be you have a space between them in your code and it's not matching exactly?

DrumClock, 2020-07-26:
Everything is the same but "echo: Active Extruder: 0" still pops up.
This is how it is written in the code:

[image]

lqbombjack, 2020-07-26:
You're missing a comma on the end of the line above. After "settings stored"

Surprising that the compiler didn't complain about that. Perhaps it just concatenated the two strings into one. That will also explain why your T1 tool change didn't echo to the screen, since it is separated by commas the string is correctly interpreted as part of the ignore list.

DrumClock, 2020-07-27:
Hi, I fixed it and it's ok.

closed 2020-08-19 by bigtreetech