bigtreetech/BIGTREETECH-TouchScreenFirmware #629: Filament motion sensor TFT display is not interested in...

AntoszHUN, 2020-04-27:
An interesting mistake. I have a filament motion sensor built from an optical switch. Marlin mode works great as it should. SKR V1.4 Turbo motherboard is connected to the E0DET connector. Flawless. He senses everything well. If it gets stuck, it gets stuck, the thread breaks it too. But. But when printing in touchscreen mode, the display firmware also ignores the sensor connected to the SKR V1.4 Turbo motherboard. I tried every possible option. It does not work when connected to the display. Only in one mode. There is filament, there is no filament. That's it. Is there a way to make the motion sensor work with the firmware of the TFT displays?

oldman4U, 2020-09-03:
Jonapot Antosz.

Is this still an issue?

AntoszHUN, 2020-09-03:
I do not know. I haven't dealt with it since. I'll check in tonight.

closed 2020-12-15 by stale[bot]

oldman4U, 2021-01-11:
Hi.

According to kisslorand this issue can be solved by adding M75 to the start gcode. This allows Marlin and therefore the mainboard to recognise that the print started which also makes the filament sensor work.

Please check and let me know

Jonapot

AntoszHUN, 2021-01-11:
> Hi.
> 
> According to kisslorand this issue can be solved by adding M75 to the start gcode. This allows Marlin and therefore the mainboard to recognise that the print started which also makes the filament sensor work.
> 
> Please check and let me know
> 
> Jonapot

I have not tried. My printer is currently being rebuilt. Maybe if you are interested check out my youtube channel. :D

oldman4U, 2021-01-13:
I checked your YouTube but my hungarian is not that good, sorry;-(

AntoszHUN, 2021-01-13:
> I checked your YouTube but my hungarian is not that good, sorry;-(

The English subtitle is already being prepared. Most videos are ready. Is it somewhat understandable?

oldman4U, 2021-01-14:
I checked the smart filament and NeoPixel movie but could not see the subtitles.

Btw, you are using the standard UI and not yours...

AntoszHUN, 2021-01-14:
> I checked the smart filament and NeoPixel movie but could not see the subtitles.
> 
> Btw, you are using the standard UI and not yours...

Those videos are not yet captioned. I'm moving a little slow with it. But I try.

oldman4U, 2021-01-14:
;-)