bigtreetech/BIGTREETECH-TouchScreenFirmware #631: BMPs NANED INCORRECTLY ACCORDING TO LIST

colinjohnm, 2020-04-27:
BMP FILES INCORRECTLY NAMED

STATUSNOZZLE
STATUSBED
STATUSFAN
 should be STATUS_NOZZLE
STATUS_BED
STATUS_FAN
 According to LIst

So Fail to Load and leave front screen incomplete

Hope this helps

closed 2020-04-30 by bigtreetech

NAPCAL, 2020-05-05:
Since the renaming these files it has caused an issue with currently compiled .bin updates because those still use the old file names. So most users end up with this. 
[image]

NAPCAL, 2020-05-05:
Since the renaming these files it has caused an issue with currently compiled .bin updates because those still use the old file names. So most users end up with this.

PR #619 (2020-04-30): fix icon names

Fixes #631 after last commit