bigtreetech/BIGTREETECH-TouchScreenFirmware #632: Two horizontal lines always present on LCD when touch screen cable is hooked up

ColdFusionShare, 2020-04-28:
### Description

Two horizontal lines always present on LCD when touch screen cable is hooked up

### Steps to reproduce

Attach touch screen cable and attache usb cable to sk4 1.4 turbo

**Expected behavior**
I expect a clear screen and graphics and not artifacts

**Actual behavior**
Artifacts over top of proper graphics and very low contrast

### Hardware Variant
box and board has no # ordered 3.5 TFT with SKR 1.4 turbo

### Additional Information
Picture attached
[image]

Email: pbauer@ict-computing.com

guruathwal, 2020-04-28:
most probably you plugged the cable reversed in SKR 1.4. If it is, then your serial pins are dead now.

ColdFusionShare, 2020-04-28:
Hello Gurmeet, 

I connected the line with 5V towards the DC in connection side and the
reset pin to the other side. 

When I unplug this connection and only have EXP1 and 2 I see no lines
and the contrast is normal. 

It is a V3.0 display. 

My testing was with both EXP1 and 2 and the serial cable plugged in so
that I could switch modes. 

I never had the cable plugged in in the opposite way. 

Is there any way to test if any damage has been done? Is this to the LCD
or the SK4 1.4 Turbo? 

Regards, 

Peter 

---

Peter Bauer, President
ICT COMPUTING, INC.
1818 Yorkshire Drive
Val Caron, Ontario P3N 1R9
Canada

TEL: 705-673-3324

 

Links:
------
[1]
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/632#issuecomment-620402210
[2]
https://github.com/

ColdFusionShare, 2020-04-28:
Hi Gurmeet, 

I double checked my connection and you are correct! I followed a video
on the 1.4, but I assume the connector 

was changed for flipped around on the SKR. 

So, which board is fried? SKR or LCD or both? How can I test to check if
this actually occurred? 

I hooked up correctly and now there are no lines. 

I cannot get past printer not connected. I've added all 5 drivers
(2209s). 

There are the configuration.h settings I have .... 

#define SERIAL_PORT 0 

/** 
 * Select a secondary serial port on the board to use for communication
with the host. 
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7] 
 */ 
#define SERIAL_PORT_2 -1 

/** 
 * This setting determines the communication speed of the printer. 
 * 
 * 250000 works in most cases, but you might try a lower speed if 
 * you commonly experience drop-outs during host printing. 
 * You may try up to 1000000 to speed up SD file transfer. 
 * 
 * :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000] 
 */ 
#define BAUDRATE 115200 

define LCD_INFO_SCREEN_STYLE 0 

//#define REPRAP_DISCOUNT_SMART_CONTROLLER // PB Changed - removed 
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER 

Hope this helps. Guess that the connection issue could be due to board
damage? 

Best regards, 

Peter 

---

Peter Bauer, President
ICT COMPUTING, INC.
1818 Yorkshire Drive
Val Caron, Ontario P3N 1R9
Canada

TEL: 705-673-3324

 

Links:
------
[1]
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/632#issuecomment-620402210
[2]
https://github.com/

guruathwal, 2020-04-29:
the TFT serial pins are fried. you can try connecting to any other serial port on the TFT

ColdFusionShare, 2020-04-29:
Hello Gurmeet, 

After flipping the connector around and setting the baud rate on the LCD
to 115k 

I now get a connection to the printer. I can run Gcodes via the terminal
and get a response. 

So, I assume this means that the serial port on LCD connector is ok. 

Thanks! 

Peter 
---

Peter Bauer, President
ICT COMPUTING, INC.
1818 Yorkshire Drive
Val Caron, Ontario P3N 1R9
Canada

TEL: 705-673-3324

 

Links:
------
[1]
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/632#issuecomment-621049620
[2]
https://github.com/

guruathwal, 2020-04-30:
Good to know that. Please close this issue.

closed 2020-04-30 by ColdFusionShare