bigtreetech/BIGTREETECH-TouchScreenFirmware #634: The TFT display does not support the progress LED.

AntoszHUN, 2020-04-28:
Is it possible that the Printer Event LEDs feature will be included in the TFT display firmware at some point? Marlin knows. It also works nicely in Marlin Mode. But not in touch screen mode anymore. Do you know anything about this?

Hello! AntoszHUN

GregSKR, 2020-05-04:
That would be great!

oldman4U, 2020-09-11:
Jonapot.

Does this cover your feature request?

traffic-light, 2020-10-11:
Copy from #1072 :

I'm currently working on a firmware change that will change the encoder knob colour when heating (while printing).
When heating is done it become the colour you did set in the feature settings
When the print is finished it will be green.

When leaving the print menu it will be, again, the colour you did set in the feature settings

I also implemented it so that the neopixel will do the same as it would when printing form marlin:
Heating: Blue to red
Printing: White
Done: Green

I will also make the colour of the knob red when PID tuning is buzzy and green when ready ;-)

I can't test my code. My screen is broken. So you've got to wait ;-)

What screen type do you use?
I could send you a binary to test my current progress.

(You can find the code here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/compare/master...traffic-light:neopixel_sequential_whileprint?expand=1

It needs some fine tuning. But I hope it works. Can't test it....)

GregSKR, 2020-10-15:
Nice work, but I cant test it, my screen is a TFT35 v2.0 without encoder knob, so I have no calls to loopDimTimer()
My NeoPixel LEDs are pluged on my SKR v1.3, I do not have encoder LEDs...

BTW, I see in your code that you're using M150 commands with "G" letter for green:
```
      //Set neopixel and ledknob to green
      storeCmd("M150 R0 G255 B0");
```
But the letter for green is "U", if I'm not wrong: https://marlinfw.org/docs/gcode/M150.html
Like there:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/dd12c39659f3ca54f82f5da4f2f6e20476f24010/TFT/src/User/Menu/MachineSettings.c#L159

traffic-light, 2020-10-15:
It's also for neopixels ;-)

Thanks for the feedback on the green neopixel command. Did not look good enough in the gcode I think

I will fix things that are not right at this moment when I've got some time and when bigtreetech decides what to do with the memory problem

oldman4U, 2020-10-20:
Hi AntoszHUN.

Could you please close this ticket. Your requested feature is under development and you could help us to keep this ticket system usable by closing this ticket.

Thank you

closed 2020-12-26 by stale[bot]