bigtreetech/BIGTREETECH-TouchScreenFirmware #637: TFT35 E3 V.30 / Ender 3 / "no printer attached"

wearebeta, 2020-04-29:
Maybe someone can give me a helping hand. I have the Ender 3 and made a 1:1 swap with the TFT. Everything ist fine and also works under merlin, but on touch mode, I always see "no printer attached" even I changed to "BAUDRATE 115200". Any hint of how to fix?

Grrby, 2020-04-29:
Wrong connection of black cable (RS232) I guess. Check with this video
https://www.youtube.com/watch?v=l1Putn10_Ek&t=811s

Rainbow (or gray) ribbon cable is for standard display mode connection AUX/CD-10 connection
Black ribbon cable is for touch mode.

wearebeta, 2020-04-30:
Do you mean this one?
https://www.dropbox.com/s/ru12l67290e89r9/ttf.png?dl=0

Grrby, 2020-04-30:
Yes this black ribbon gable is for rs232 communication (touch mode). There no good instruction of correct connection so it is easy to connect this cable wrong,  especially on the other side, at the mainboard.

wearebeta, 2020-04-30:
Thanks for your support. I have at the moment the "Creality3D Mainboard Version 1.1.5" for Ender-3. I am not sure where to plug the black cable?
Pic: https://cdn.shopify.com/s/files/1/0217/6004/9216/products/c55bdc36d3_1024x1024@2x.jpg

GhostlyCrowd, 2020-04-30:
> Creality3D Mainboard Version 1.1.5

you cant, that board doesnt have tft output (a second serial port)

wearebeta, 2020-04-30:
Thank you! So have to wait for the "BIGTREETECH SKR Mini E3" ;)

closed 2020-05-01 by wearebeta

Ripper80, 2020-08-16:
> Maybe someone can give me a helping hand. I have the Ender 3 and made a 1:1 swap with the TFT. Everything ist fine and also works under merlin, but on touch mode, I always see "no printer attached" even I changed to "BAUDRATE 115200". Any hint of how to fix?

I have the same exact issue.  Thinking I messed up the board(s), I have replaced the SKR 1.4 Turbo, the TFT35 V3.0, replaced all the wiring that I added for dual 5015 fans, compiled Marlin standard, bugfix, and Cheetah...all with exactly the same NO PRINTER ATTACHED.  I have reversed the wires and used Teaching Tech and other videos to orient the cables...NO GO.  Baudrate for both MB and TS were changed to 115200 and 250000, same message.  People recommended I use the CR10 setup, no go....replaced all the EXP cables from 4 different companies...all with same effect....I am running out of options.  It works (I THINK) in Marlin simulation, but whats the sense of have paid for a touchscreen if that mode doesn't function???

heartfire567, 2020-09-09:
> I have the same exact issue. Thinking I messed up the board(s), I have replaced the SKR 1.4 Turbo, the TFT35 V3.0, replaced all the wiring that I added for dual 5015 fans, compiled Marlin standard, [...]

same issue as well marlin works for me and i can print but the tft wont connect to the board in btt mode

phongshader, 2020-09-25:
I'm having the came issue, exchanged tft and the new one had the same issue. Attaching it to the tft pins on a skr mini E3. Are there settings in Marlin that need to be set?

IUbique, 2020-10-18:
You probably connected the cables wrong and the rx in uart on tft35 was damaged. ...that's what I did. Fortunately, the TFT35 E3 has two additional UART channels.

You only need to compile the modified version of the firmware and connect the RX and TX cables to a different UART channel than originally intended.

Before compiling the latest version, you should change the settings in the file
BIGTREETECH-TouchScreenFirmware \ TFT \ src \ User \ Variants \ pin_TFT35_V3_0.h in the line responsible for SERIAL_PORT settings.
For me after the change it looks like this:
`/*
 * SERIAL_PORT: communicating with host(Marlin, smoothieware, etc...)
 * SERIAL_PORT_X: communicating with other controller(Octoprint, ESP3D, other UART Touch Screen, etc...)
 */
#define SERIAL_PORT   _USART3  //default usart port
#define SERIAL_PORT_2 _USART1
#define SERIAL_PORT_3 _USART2
#define SERIAL_PORT_4 _UART4`

After compiling and loading the new firmware to our device, it remains to slightly change the cables. + 5V and -GND power remains where it was. However, RX and TX are connected to the new UART. In my case, UART3.

Now everything will work again. Including touch mode. You just need to carefully check where we connect what.
[image]

soptik4542, 2020-11-10:
hi.IUbique
I could ask you for help.
could you please upload your compiled bin for the previous post. TFT35v3 with your modified UART port.
You would help me a lot, I haven't done a debug yet.
Thanks a lot...

soptik4542, 2020-11-10:
> You probably connected the cables wrong and the rx in uart on tft35 was damaged. ...that's what I did. Fortunately, the TFT35 E3 has two additional UART channels.
>
> You only need to compile the [...]

hi.IUbique
I could ask you for help.
could you please upload your compiled bin for the previous post. TFT35v3 with your modified UART port.
You would help me a lot, I haven't done a debug yet.
Thanks a lot...

soptik4542, 2020-11-11:
solved ... I bit through copying ..