bigtreetech/BIGTREETECH-TouchScreenFirmware #639: USB disks are not loaded - for guruathwal

radek8, 2020-04-30:
for @guruathwal :
In bigtreetech/BIGTREETECH-TouchScreenFirmware#627
You have an error.
USB disks are not loaded after editing
U_DISK_SUPPROT is not U_DISK_SUPPORT

After overwriting to U_DISK_SUPPROT it is functional.
I also stopped reading SD on the board. I did not find an error.
Can you check it?
Thank you
[image]
[image]
[image]

radek8, 2020-04-30:
File Print.c

Balzac40, 2020-05-01:
I confirm !

guruathwal, 2020-05-01:
@radek8 thanks for Identifying. I think `U_DISK_SUPPROT` was misspelt by someone at the time of adding it and I unknowingly used used the correct spelling which resulted in the bug.

radek8, 2020-05-02:
@guruathwal 
Thank you
You are doing a good job

PR #641 (2020-05-05): Fix Icons & fix misspelt U_DISK_SUPPROT bug 
- Rename missing icons for TFT24 & TFT28 not renamed through #619
- Fix misplaced new icons for 'percentage'
- Fix misspelt U_DISK_SUPPROT to U_DISK_SUPPORT
- Fix Rotate UI on reset and rename Reset file only after completing reset process
Resolves #639, resolves #638
closed 2020-05-05 by bigtreetech