bigtreetech/BIGTREETECH-TouchScreenFirmware #642: Communication timeout - reset send buffer block

QuitButton, 2020-05-01:
### Description

Just recently I'm seeing lots of "Communication timeout - reset send buffer block" errors from Repetier Host when printing using USB and having the TFT in Touch mode. This error doesn't happen with the TFT in Marlin mode.

I also notice there's lots of visual stuttering and pausing with Repetier as it visualises the printout while printing, suggesting there's buffer blocking happening more often that just when the error message shows.

I've tried changing baud rate, increasing various serial buffer sizes in Marlin, but so far change has made an obvious difference.

### Steps to reproduce

1. Print something using Repetier host via USB 

**Expected behavior**
No errors.

**Actual behavior**
"Communication timeout - reset send buffer block" on a regular but unpredictable regularity 

### Hardware Variant
TFT24 V1.1

zonalimitatore, 2020-06-23:
Do you solved? I've a random communication errors with Marlin's mode too

QuitButton, 2020-06-23:
Nope. It happens randomly - the hardware seems to get into a mess and will result in errors and failed prints at times. Powering off and on again and it'll fix itself and be fine for days until it gets in a mess again. For me it will occasionally do this in Marlin mode on the TFT, but its quite a regular thing when in touch screen mode.

I've tried playing with the buffer sizes etc. in Marlin and it made no obvious difference.

There's obviously some timing issue with the USB, I'm guessing its similar to a lot of the USB-Serial issues that where the bane of some 8Bit 3D printer boards, but I've got no way of diagnosing exactly where the issue lays.

zonalimitatore, 2020-06-23:
Tomorrow i try to print in touch mode.my baud is 115200

zonalimitatore, 2020-06-24:
Ok...I've print all day in touch mode,using usb cable and repetier.No message about connection errors or resend.
I've do a change in buffer (config_adv) and activate serial xon xoff

oldman4U, 2020-10-09:
What is the status of the ticket?

QuitButton, 2020-10-16:
> 
> 
> What is the status of the ticket?

Ignored?

oldman4U, 2020-10-16:
Ok. 

Have you tried the changes @zonalimitatore has suggested? Or anything else?

closed 2020-12-22 by stale[bot]

oldman4U, 2021-01-11:
Hi.

With the latest TFT firmware, some buffers have been changed. Could you please check if this solves the problem you reported and let us know.

Thank you

QuitButton, 2021-01-12:
"latest" firmware being released on 11 Apr 2020?

oldman4U, 2021-01-12:
Latest firmware being released today.

oldman4U, 2021-01-14:
Have you found the firmware and uploaded it to your TFT?

QuitButton, 2021-01-15:
No, where is this "latest release"? The latest I can see is Vx.x.26 Latest on 11 Apr 2020

JerksonS, 2021-02-13:
I too have this problem on a biquB1. Flashed with the latest screen firmware from here 2 days ago. Printing from usb/sd card is fine, but repetier ends up with this same error randomly (usually within the first 1 hour of a print)

oldman4U, 2021-03-08:
You can find the firmware from today here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware.git

oldman4U, 2021-03-25:
@ghost @JerksonS

Could you please let me know the status of the reported issue.

Thank you

oldman4U, 2021-03-25:
Seems the user does not exist anymore. So it is about you @JerksonS. Have you been able to download the latest firmware and upload it to your TFT?

JerksonS, 2021-03-26:
The issue seems to have self-resolved. I cant recall if I flashed since I last reported the issue. Let's consider it resolved :)

oldman4U, 2021-03-26:
Thank you

Happy printing