bigtreetech/BIGTREETECH-TouchScreenFirmware #644: [BUG] 🔋 BTT_UPS_24V problems with BTT_TFT3.5_E3 (PLR)

TheMiguelBi, 2020-05-01:
### Description

Good morning,
I have been trying for several days to operate the **_Powerloss_Recovery_** function using the Bigtreetech 24 volt UPS on my CoreXY custom printer.
But this does not behave regularly using the Graphic Display connected via RS232. Even my NeoPixel LEDs don't work 
(With simulation LCD display .... everything smooth! NeoPixel They also work!! 🧐)

### Steps to reproduce

1. I start a print via SD TFT display _(The PLR on onboardSD does not work using TFT_mode🤷🏻‍♂️)._
2. After a few layers I simulate a power outage.
3. I restore the power source and ....

**Expected behavior**
I expect an automatic screen asking for the will to recover the interrupted print.
And to be able to save 3D printing!

**Actual behavior**
The display shows the "HOME" screen. End.
If I move to "Print"->"SD TFT" then the print recovery window appears.
If I select "Resume" the plate and extruder are brought to temperature. Subsequently, homing of the X and Y axes, extrude Xmm of the filament and going to the XY point of printing interruption, but the **Z coordinates are incorrect**. The print is "far" from the extruder.

### Hardware Variant
I use a **BTT GTR V1.0** card and a **TFT3.5 E3 V3.0** display.
**BL-Touch v3.1** for plate leveling and **BTT_UPS_24v** for backup. 
### TFT Firmware Version & Main Board Firmware details
Current TFT Firmware version BIGTREE_TFT35_V3.0.26.bin & Marlin firmware version bugfix-2.0.X updated on 20/04/2020. 

### Additional Information
By performing the same steps with **LCD Mode Display**, everything **WORKS** smoothly!
When the power is restored, the screen to recover the interrupted print automatically appears (see attached photo).
P.S. A curious thing happens, Subsequently the steps are different from the Graphic Display mode:
First homing X and Y axes, waits for the bed and extruder temperature, extrudes Xmm of material and moves to the last coordinates saved before the interruption, correctly restoring printing. Z axis is exactly in the correct point.
[image]
- Using the TFT display in the **LCD simulation mode**, I can regularly print from the SD display (or from the onboard microSD if enabled in the firmware). After an electrical outage I can recover the print correctly. My NeoPixel LEDs work before, during and after printing. All is perfect.
- Using the TFT display in **Graphic mode** nothing works!
The onboard microSD card is read, the PLR file is created during the interruption, but when the power is turned back on, there is no possibility to recover the print. With SD on the display it is possible to manually recover the print (by entering the menu print-> SD), but the Z axis coordinates are not reset correctly. So the 3DPrinting will be lost.

🤕 **I do not know what to do. Anyone know the solution?**
Thanks!

oldman4U, 2020-09-03:
Hi.

There have been many fixes in Marlin and TFT SW in the meantime. Have you installed them and what is the status of the reported issue?

Please let us know

oldman4U, 2020-10-09:
Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2021-01-05 by stale[bot]

DaStivi, 2021-02-04:
i've to reopen this ticket as its same issue for me... its exactly as decribed from the creator of the issue... when poweroutage occurs, and after resume the print from the print/sd-udisk menu (what wouldn't be the issue for me todo this) x and Y homes, then the print head travels back and subsequently moves down on z until it crashed into bed (when the print wasn't high enough..)  ultimately it doesn't raise on poweroutage, but still travels down the required Z height...