bigtreetech/BIGTREETECH-TouchScreenFirmware #645: [BUG] unable to revert z motor direction

Cadaxe20000, 2020-05-02:
After installing new trt firmware I am unable to revert z direction under the machine configuration. That function change the order of the buttons and nothing else

oldman4U, 2020-05-03:
Hello Cadaxe20000.

I believe what you describe as a bug works exactly as intended. You will have to invert the movement of the axis in the firmware of your mainboard in Configuarion.h:

// Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
#define INVERT_X_DIR false or true
#define INVERT_Y_DIR false or true
#define INVERT_Z_DIR false or true

Hope this helps

oldman4U, 2020-05-04:
Hi.

Have you had time to check your firmware in the meantime?

Please let me know.

oldman4U, 2020-06-26:
hi.

what is the status of the reported issue. It would be great, if you could close the ticket in case the issue is solved.

Thank you

closed 2020-07-04 by bigtreetech