bigtreetech/BIGTREETECH-TouchScreenFirmware #649: [FR] Auto off screen

davidcgu, 2020-05-04:
Description
Have the option to enable that screen turn off after certain amount of seconds with no touch as it use to do any other screen, maye dwell time can be configurable and feaure to have some kind of enable/disable selector.

Benefits
Durablility on the screen, energy saving, light off at night.

Related Issues
N/A

oldman4U, 2020-05-04:
HI davidcgu.

I use the the dim function which comes with the latest firmware on my TFT35 and this allows me to reduce the brightness down to 5% with the ability to define the full brightness time. This works very well.

davidcgu, 2020-05-04:
Thanks for your comments, maybe I missed that option as didn't saw that, will take a look and let you know.

Regards

oldman4U, 2020-05-04:
You are welcome. Good you asked;-)

The function came with the latest firmware: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/releases

Please close this ticket if it works for you.

davidcgu, 2020-05-04:
That is great however  unfortunately my TFT is V2.0, so not available for now:

(only TFT24 V1.1, TFT35 V3.0, TFT35 E3 V3.0, TFT28 V3.0, supported)

Regards

radek8, 2020-05-04:
V 2.0 does not have compatible HW

closed 2020-05-05 by davidcgu