bigtreetech/BIGTREETECH-TouchScreenFirmware #655: Onboard Sd print not working even after enabling it in settings

tanellaanemagi, 2020-05-05:
Onboard Sd print not showing up even after enabling it in settings. also send cancel g code not working.

radek8, 2020-05-05:
91/5000
You need to download the latest Marlin 2.0 with bug fixes, to support printing from SD oboard

tanellaanemagi, 2020-05-05:
i have skr mini 1.2 with marlin 2.0 and it worked with previous firmware because there was no   "infoMachineSettings.onboard_sd_support"  it seems like this code doesent read this value or something with that, same problem is with   "infoSettings.send_cancel_gcode "
in previous version this also worked.

radek8, 2020-05-05:
Yes you're right.
It used to work, now you have to upload a new MARLIN

Look at CONFIGURATION.H, line 240 :-)

Starting from Marlin Bugfix 2.0.x Distribution Date: 2020-04-27 & above, The TFT will auto detect

rozenrod, 2020-05-05:
I downloaded the version of marlin on 05/03/20, in the Configuration_adv.h on line 1209 I registered #define SDCARD_CONNECTION ONBOARD, I reshooted it, but the screen displays an error M21

tanellaanemagi, 2020-05-05:
i´m using marlin 2.0.5.3

tanellaanemagi, 2020-05-05:
[image]
 this seemed to me to be in wrong place

radek8, 2020-05-05:
> Verze marlin jsem stáhl dne 05/03/2012, v Configuration_adv.h na lince 1209 jsem zaregistroval #define SDCARD_CONNECTION ONBOARD, změnil jsem to, ale na obrazovce se objeví chyba M21

2012 :-) 
You downloaded Marlin - Latest release, or Marlin 2.0 with bug fixes?
You have to download Marlin 2.0 with bug fixes

tanellaanemagi, 2020-05-05:
I downloaded latest from bigtreetech for skr mini 1.2 e3.

radek8, 2020-05-05:
> [image]
> zdálo se mi, že je na špatném místě, komentoval jsem to a dal tam, kde si myslím, že by to mělo být.
> pak jsem dostal toto místo, kde by měl být tisk sd, ale žádná ikona tam,

download new Marlin 2.0 with bug fixes

rozenrod, 2020-05-05:
> > Verze marlin jsem stáhl dne 05/03/2012, v Configuration_adv.h na lince 1209 jsem zaregistroval #define SDCARD_CONNECTION ONBOARD, změnil jsem to, ale na obrazovce se objeví chyba M21
> 
> 2012 :-)
> Вы скачали Marlin - Последний выпуск или Marlin 2.0 с исправлениями ошибок?
> Вы должны скачать Marlin 2.0 с исправлениями ошибок

[image]

I downloaded the latest Marlin Bugfix 2.0.x, it was updated 13 hours ago

radek8, 2020-05-05:
You need to enable LONG_FILENAME_HOST_SUPPORT in Marlin. 
Configuration_adv.h file

rozenrod, 2020-05-05:
> Вам необходимо включить LONG_FILENAME_HOST_SUPPORT в Марлине.
> Файл config_adv.h

Did not help

radek8, 2020-05-05:
You have enable  #define AUTO_REPORT_SD_STATUS ?

rozenrod, 2020-05-05:
> Вы включили #define AUTO_REPORT_SD_STATUS?
Yes.
I used to turn on #define SD CARD CONNECTOR ONBOARD and my SD card didn’t open on my PC, I had to enter the M22 command, but now I get an error with the M21 and M22 commands

rozenrod, 2020-05-05:
Here are my configurations
[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4583759/Marlin.zip)

radek8, 2020-05-05:
Stupid question, but I'll ask.
Do you have #define SDSUPPORT enabled in Configuration.h? :-)

[image]

rozenrod, 2020-05-05:
it's just awful, with the new Marlin configuration, I forgot to uncomment #define SDSUPPORT!
Thanks a lot, now everything works)

rozenrod, 2020-05-05:
Is there a way to upload files to an SD screen or board via Wi-Fi?

tanellaanemagi, 2020-05-06:
i got this working also, i had latest regular build not bugfix.

radek8, 2020-05-06:
> Is there a way to upload files to an SD screen or board via Wi-Fi?

It should work
I can try it tonight

radek8, 2020-05-06:
> Existuje způsob, jak nahrávat soubory na obrazovku SD nebo na kartu přes Wi-Fi?

Copy ended in error, wait for merge
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/648

oldman4U, 2020-09-14:
648 has been added 4 months ago. What is the status of the reported issue?

closed 2020-12-15 by stale[bot]