bigtreetech/BIGTREETECH-TouchScreenFirmware #656: Immposible to navigate

statictrauma, 2020-05-06:
I am not a programmer.  It would be helpful to see languages in the title of the firmware.

I am trying to update my TFT 3.5 on my Ender 3 with am SKR Mini 1.2 and I keep getting some Chinese crap on my screen and I can't read it.  A simple addition in the title would be nice.  Now I can't read my screen and still can't find the damn update or correct firmware.

radek8, 2020-05-06:
If you can't compile yourself, here you have compiled

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update%20-%20Unified%20Menu%20Material%20theme

radek8, 2020-05-06:
What Display Version do you have? What files did you use to update?
The installation is multilingual
BLTouch will not work unless you configure Marlin.
If you don't understand, don't go for it.
Do you have a friend who understands that a little?

statictrauma, 2020-05-06:
I dont know anyone who understands code.

Version 3.0

I got it back to English finally but I had found an update for the board to
get the BLTouch to work. I had to send that kit back because it was damaged
when it got here. BTT couldn't figure it out and they asked me to return
theast one.

I just can't find the same version of firmware now.

guruathwal, 2020-05-09:
you can always reset the settings by creating a text file with name: `reset.txt` in SD card.

statictrauma, 2020-05-09:
I finally got it to work and now my BLTouch is bad. Gonna have to return it
and Uninstall it.

Thanks for y'alls help.

guruathwal, 2020-05-10:
@statictrauma good. you can close this issue now.

statictrauma, 2020-05-10:
Oh. Didnt know I can do that. Thank you.

closed 2020-05-10 by statictrauma