bigtreetech/BIGTREETECH-TouchScreenFirmware #660: [BUG] TFT35 E3 - Unified Menu Material Theme Issue - v26.1

Solidsnakey, 2020-05-07:
**TLDR: Copy all bmp files from v25 bmp folder into v26.1 bmp folder, DO NOT replacing any files with the same name.**

**Description**
When updating firmware on TFT35 E3 Unified Menu Material Theme is missing bmp files.

**Steps to reproduce**
1. Download v26.1
2. Copy folder "TFT35" and "BIGTREE_TFT35_V3.0_E3.26.1.bin" to SD card
3. Plug SD into screen and boot

**Expected behavior**
All images to display
[image]

**Actual behavior**
Error during install:
Error: Missing images for Tool, Bed and Fan0
[image]

**Hardware Variant**
Ender 3 with SKR v1.4Turbo and TFT35 E3

**TFT Firmware Version & Main Board Firmware details**
Marlin 2.0.x Bugfixes
TFT v26.1

**Additional Information**
None

**Fix**
Copied all bmp files from v25 bmp folder into v26.1 bmp folder, not replacing any files with the same name.

WubbuW, 2020-05-07:
Just wrong name of the images... remove the underscore from eht 3 status images and will be ok!

oldman4U, 2020-06-15:
Hello Neidraidd.

I hope you have been able to solve the reported issue in the meantime . So please close this ticket if you do not need it anymore.

Thank you

closed 2020-06-15 by Neidraidd